luizribeiro/mariner #25: Settings screen

luizribeiro, 2020-09-27:
Things that could be added:

* Reboot printer
* Manual control of the Z axis
* Bed leveling tool
* Printer settings such as when the fan should turn on, screensaver, etc (things that are generally available only on the gcode file that comes with the printer)
* Any information about the printer itself?
* Mariner cache settings (clear cache, bootstrap, etc)
* Mariner [app version](https://github.com/python-poetry/poetry/issues/273)

dadwarf, 2020-10-13:
`M8030 Ix` (fan : 0 off, -1 always on, -2 on when printing only)

doucettom, 2021-04-19:
This would be more of a printer control screen. A settings screen would IMHO include Title, Printer's name, Port, Color Scheme, User etc. I would probably be able to to some work but I tried to open main.js and it is not indented (not sure right word...) So I cannot read it much !