luizribeiro/mariner #56: Writing files to USB from serial

dadwarf, 2020-10-11:
Hi,
I'm trying to write files from serial on my Elegoo Mars (model without ethernet port), but i'm stuck, i start to think that it's not possible (https://www.elegoomars.com/forum/showthread.php?tid=150&pid=1575#pid1575). 
I managed to dump the firmware from the mainboard to explore it.. but it's too difficult for me..
Did you also try ?
M28 command don't act like Elegoo Mars Pro 
[image]

luizribeiro, 2020-10-11:
Hey! I'm excited you found this repo 😄 

I'm cleaning it up to hopefully make a first release soon. This is basically a Web UI for controlling the Elegoo Mars Pro. We could try to get it to work with the Mars if you're interested to help with that.

---

To your question: I actually haven't tried uploading with `M28` even on the Mars Pro.

I've been using Linux's USB Gadget Mode through the `g_mass_storage` module to share files with my Mars Pro' mainboard through the USB port.

I've found that to be a bit easier to work with, since I can read/write files from the Raspberry Pi itself easily with Python (which has allowed me to upload files remotely with a Web UI).

Have you tried using USB Gadget Mode? I'm planning on writing a tutorial (let me know if that would be helpful), but [Adafruit's tutorial](https://learn.adafruit.com/turning-your-raspberry-pi-zero-into-a-usb-gadget/overview) on the Pi Zero USB Gadget Mode is a good start.

dadwarf, 2020-10-11:
You're right it could be solve with a Pi Zero and USB Gadget mode and probably no one would like to control an Elegoo Mars from USB (to serial) but  more by the network.. that's because g_mass_storage is a trick for me more than a solution : you have to remount the share folder at every file's change. Someone has made a post about that [here}(elegoomars.com/forum/showthread.php?tid=399) & Python script [here](https://gist.githubusercontent.com/davidhoness/0f45ef6a41bac6311614f109acbf92db/raw/970badd0ae4b097e3af8d5142e65c34b21f5cfab/usb_share.py). In fact a Zero Pi will be in my Elegoo Mars .. but i'd rather if the oly storage is the USB flash drive. And yes, of course if i could help.

The Pro you have is ethernet capable ?

luizribeiro, 2020-10-11:
> you have to remount the share folder at every file's change

You don't actually have to remount the share folder on every file change. At least I don't have to with my setup. I've updated the `README.md` of this repo with some rough instructions on the setup I have.

> The Pro you have is ethernet capable ?

No, not at all. I think no Pros have ethernet. I'm just using the serial port like you are. The only difference is I don't use the serial port for uploading files (since I don't see the need for it).

I also keep my Pi Zero W within the printer itself and power it through the printer's power supply, so the entire setup feels super clean.

dadwarf, 2020-10-11:
Wow great !! Thanks a lot Ruiz, that's perfect and your UI is really nice.. you just convince me !

dadwarf, 2020-10-11:
You have been faster than me : i went to put a link to your Github on Elegoo's forum and you had already done it :)

dadwarf, 2020-10-11:
sudo apt update && sudo apt install libxslt1.1

luizribeiro, 2020-10-11:
haha :) Let me know if you are able to get it running on the original Elegoo Mars. Happy to make any changes to the code if needed to support it properly

> sudo apt update && sudo apt install libxslt1.1

Ah yes. There's a few dependencies for the `.deb` that need to be installed first. Should I update the `README.md` with those?

luizribeiro, 2020-10-11:
Ah, actually there's at least one feature we need to add support for to get the Elegoo Mars to work: `.cbddlp` files.

Let me file an issue for that, maybe I'll have some time for it later tonight or next weekend.

dadwarf, 2020-10-12:
Yes for me that's the only dependencie to make the deb install.. After the changes in cmdline.txt and config.txt the pi is stuck during the boot after SD mount.. strange

luizribeiro, 2020-10-12:
That sounds about right. I'm using buster too.

What does your `cmdline.txt` look like? Here's mine after all changes:

```
console=tty1 root=PARTUUID=18b5dae3-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait modules-load=dwc2
```

The install instructions I wrote on the repo are also trying to do everything at once. That may be a bit too much. Maybe you can try setting things up one by one to see where the system becomes unusable?

dadwarf, 2020-10-12:
https://www.raspberrypi.org/forums/viewtopic.php?t=264296
That's a clue, no ?

you're right :

1. First boot OK
2. marina3d install & reboot OK
3. dtoverlay=dwc2 > /boot/config.txt  & dwc2 > /etc/modules (for testing) & reboot OK
4. piusb.bin & fstab & reboot KO

Stuck during piusb.bin's mount because it is not formated, no ?

luizribeiro, 2020-10-12:
Oh yeah you're totally right. I forgot that on the instructions! Does that fix the issue you're running into?

dadwarf, 2020-10-12:
It would be nice to identify the printer model by the firmware version (and also showing the version) with `M4002`

dadwarf, 2020-10-12:
I will try a print later (the mainboard is on my desk for testing), but why would it be different : the GCODE you use are common between pro and non pro.. and i could generate and print CTB files on my Elegoo Mars !

luizribeiro, 2020-10-12:
> It would be nice to identify the printer model by the firmware version (and also showing the version) with `M4002`

Is that possible? My firmware version on the Mars Pro is `V4.3.4_LCDC` (according to `M4002`).

`M115` (which supposedly returns when the firmware was built) gives me: `CBD make it.Date:Apr  2 2020 Time:16:30:35`

Maybe we can map firmware versions to printers to detect them? Filed #64 to look into that. Feel free to add your own observations from the original Mars to that issue.

> i could generate and print CTB files on my Elegoo Mars !

I'm not sure if that's possible, as far as I know the Elegoo Mars only supports `.cdbdlp`? Unless one of the more recent firmwares adds supports for `.ctb` too....

If `.ctb` doesn't work, I think you *might* be able to get it to print by doing this:

1. Upload a `.cdbdlp`
2. Open it on the file browser from mariner
3. The file details won't show (we only support `.ctb` files for now)
4. You *might* still be able to hit print regardless of that. Print status might be super wonky/broken because we won't be able to read the `.cdbdlp` metadata to know which layer is currently being printed

dadwarf, 2020-10-13:
According to the Elegoo Forum there are 3 version of Mars : ORIGINAL with ethernet port, STANDARD and PRO have the same mainboard.

- Original up to V4.2.20.3_LCDM
- Normal up to 4.3.4_LCD(F/M/C?)
- Pro up to 4.3.4_LCDC

So no, there's no way to identify model by version.. 

We have the same firmware / mainboard so for the CTB that ok

dadwarf, 2020-10-13:
I had upgraded my Elegoo Mars to v4.3.4 (from v4.3.2) to match yours : First i tried to upload the 2 files (update.SH4, ALFPGA213_SPI_MIPI_P316_CTB_1440P_140M_FF.CBD) with the UPLOAD button of Mariner3D but your filter disable that.. so i SCP the 2 files to /mnt/usb_share.. then (i'm scare of nothing) i print the 2 files (that's the upgrade method) .. and now i'm also in v4.3.4. But i had to delete the 2 files before beeing able to list the file content from Mariner3D.
[image]

luizribeiro, 2020-10-17:
> But i had to delete the 2 files before beeing able to list the file content from Mariner3D.

Ahh, good catch. I filed #74 to fix that.

luizribeiro, 2020-10-17:
@dadwarf, going to close this issue since I think we were able to get this working on your printer? Feel free to open more issues if there's more things you would like to see or if you run into more problems!

closed 2020-10-17 by luizribeiro