luizribeiro/mariner #130: [INFO] Works with Mars 2 Pro

gschmidl, 2020-11-23:
Not a bug, just confirming that mariner works fine with the Mars 2 Pro, though I cannot recommend putting the Pi Zero inside the printer as the wifi is not strong enough for a reliable connection.

luizribeiro, 2020-11-24:
Hey @gschmidl! Thanks for reporting back! I'll keep this issue open until I update the documentation to reflect that the Mars 2 Pro works with mariner.

dogmachine, 2020-12-23:
Also works with Mars 2 non-pro version.

I put it inside the chassis, seems fine, unifi access point reporting a strong signal.

Additionally, I have powered my pi0w from the 5v & gnd pins on the serial headers.  I had already done this before finding your project, so adding two more jumper wires made it a trivial upgrade.

rkolbi, 2020-12-27:
Great project luizribeiro! Thank you very much for all the effort!

I have it in a Mars2Pro as well. I added a smb share, then mapped out with drive letter on my pc. This makes it much easier as then you just save to the samba share location with the slicer and start print - vice saving locally and then uploading to website.

commit 8dae4417 (2021-01-09): Updated README.md with supported printers

Fixes #130
closed 2021-01-09 by luizribeiro