luizribeiro/mariner #152: '//Disk init fail!\r\n'

Forions31, 2020-12-03:
Hello,
I have a problem. When I am in the home of Mariner all is well finally I think.
[image]
when i am in the file tab no problem either i have this.
[image]
but when i press print i get this error message.
[image]
I'm not sure where I made a mistake. I show you my files

Config.txt

[image]
[image]

cmdline.txt

[image]

fstab

[image]

rc.local
[image]

The firmware of my elegoo mars pro is 4.3.4
Thanks in advance for your help and sorry for my english :-)

luizribeiro, 2020-12-04:
Hi! Thanks for reaching out. Your printer is not recognizing the Pi as a USB mass storage device. But everything else seems to be fine.

Can you try moving these lines of your `/boot/config.txt` under the `[all]` section like this?

```
[all]
dtoverlay=dwc2
enable_uart=1
```

Let me know if this fixes the issue! I should probably clarify that under the instructions.

luizribeiro, 2020-12-04:
There's actually another issue on your `rc.local`. This is what mine looks like:

```
#!/bin/sh -e

_IP=$(hostname -I) || true
if [ "$_IP" ]; then
  printf "My IP address is %s\n" "$_IP"
fi

modprobe g_mass_storage file=/piusb.bin stall=0 ro=1

exit 0
```

Note that you have an `exit 0` just before the `modprobe` command on yours, which will cause the script to end execution before the `modprobe` command runs. Just removing that should be enough (you also don't need the additional `#!/bin/sh -e` later on in the file, but there shouldn't be any harm in that).

Forions31, 2020-12-04:
Hello,
Thank you for your fast answers.
but i have always the fail message.
[image]
for rc.local i copy/past your answers.
[image]
in doubt I also made a total installation of Mariner but i have the same message.

[image]
[image]

luizribeiro, 2020-12-05:
The order of your `fstab` matters. Can you try moving the `/piusb.bin` line to the end? Make sure to reboot your Pi after doing that.

Some other questions that may help debug:

1. Can your printer see the Raspberry Pi as a USB device? Have you plugged it into the printer's USB port?
2. Does `/piusb.bin` exist? Is it mounted properly? You can check if it is mounted properly with `df -h`. `/mnt/usb_share` should show up there as `/dev/loop0` or something similar.
3. You can also run `dmesg` as root to see if there are any interesting kernel messages that will help debugging.

Another thing you could do is just try to get the RPi to work as a USB mass storage gadget separately against a computer or something that is a bit easier to debug. There's a bunch of tutorials online for that.

Forions31, 2020-12-06:
Hello,
yes I have it connected to the printer. But she can't see anything.
I just changed the position of the "piusb" line but still nothing.
[image]

When I enter the line ... I have this message which seems good to me.
[image]

when i run sudo dmesg
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Forions31, 2020-12-07:
subject closed error on my part sorry

luizribeiro, 2020-12-12:
Hey @Forions31, I'm running out of suggestions unfortunately :( I can't spot anything weird on your configs myself.

This is something about the way that your USB Mass Storage gadget is setup on the Raspberry Pi.

For what it's worth, these are the messages I see on `dmesg` upon a successful boot with the USB mounted properly:

```
[   36.029420] Mass Storage Function, version: 2009/09/11
[   36.029445] LUN: removable file: (no medium)
[   36.029622] LUN: read only file: /piusb.bin
[   36.029636] Number of LUNs=1
[   36.042125] g_mass_storage gadget: Mass Storage Gadget, version: 2009/09/11
[   36.042147] g_mass_storage gadget: userspace failed to provide iSerialNumber
[   36.042158] g_mass_storage gadget: g_mass_storage ready
[   36.042177] dwc2 20980000.usb: bound driver g_mass_storage
[   37.347181] dwc2 20980000.usb: new device is full-speed
[   37.510993] dwc2 20980000.usb: new device is full-speed
[   37.549309] dwc2 20980000.usb: new address 1
[   37.588413] g_mass_storage gadget: full-speed config #1: Linux File-Backed Storage
```

So it looks fairly similar to yours.

My suggestion here to debug this further would be:

1. Make sure your `/piusb.bin` was formatted as FAT32 (i.e. `mkdosfs /piusb.bin -F 32 -I`). I know that the Mars has trouble reading anything else. I don't think this is the problem here, since you're mounting it as `vfat` - unless you formatted it as FAT16 or something accidentally. Maybe try creating a brand new `/piusb.bin`?
2. Upgrade everything just in case?
3. Try reading the Raspberry Pi's USB storage with a device that isn't your Mars printer?

For 3/, I would even suggest to go as far as following a tutorial such as [this one](https://magpi.raspberrypi.org/articles/pi-zero-w-smart-usb-flash-drive) from MagPi. The problem really shouldn't be mariner if your printer isn't being able to see the Pi as a USB storage device. It seems like mariner can even communicate with the printer through the serial port - so everything on your setup looks fine.

Forions31, 2020-12-12:
Hello,

It's good I succeeded !!!!!!

I changed the usb cable 3 times and made some modification in the lines.

my config.txt
[image]

cmdline.txt
[image]

fstab
[image]

rc.local
[image]

I don't know what the correct modification is but it works lol.

I have just this message but is not a fatal error
[image]

Just last question is it possible to integrate mariner in home assistant ? 

Thank you for your help :-D

luizribeiro, 2020-12-18:
> Just last question is it possible to integrate mariner in home assistant ?

Yup, it definitely would be! I use Home Assistant myself, but haven't found a use to have my 3D printer exposed on it. Feel free to create an issue for it if this is a feature you would like to see.

Going to close this issue since your issue seems to have been resolved.

closed 2020-12-18 by luizribeiro