luizribeiro/mariner #160: Print percent and layer in title

cbusillo, 2020-12-12:
I love this server.  It works perfectly.  It would be great if I could see the percentage complete and layer level.

As a separate comment, do you have issues with wifi reception?  When I put the back plate on, my reception drops too low.  Have you seen a plastic back plate to print to replace the metal one?

Thank you for the great work.  I'm looking forward to seeing where this project goes.

luizribeiro, 2020-12-12:
That's a good idea! I think this should be easy to implement. I may put just the percentage on the title for simplicity, but I don't feel too strongly about it. Let's see how it feels :)

> As a separate comment, do you have issues with wifi reception?

I haven't had any, no. It will depend a lot on your router, how far you are from it, etc. I'm just a couple of walls away from my router, so I haven't had any issues. Someone mentioned on another thread similar issues with leaving the Pi inside the printer case. I guess you could also consider leaving it outside and just routing the wires through the case's holes?

cbusillo, 2020-12-12:
I have your source version installed, so I’ll try it whenever you update it. 

I’m quite a distance from my router. Good suggestion on routing the cables out of the housing. I hadn’t thought of it since it’s so neat inside the machine. Neat and not working ;)

cbusillo, 2020-12-12:
This is super easy to do.  You may want to include this in your blog post.  The housing of the Mars is all metal and reduces the signal strength by quite a bit.  Let me know if I can do anything to help out with your project.

https://www.briandorey.com/post/raspberry-pi-zero-w-external-antenna-mod

juicyroots, 2021-04-30:
It would be great to have some more information in the browser tab titles / page: like printer name, filename, current layer, percentage complete, status, etc. Maybe even the ability to choose with checkboxes.

This would be cool...
SM4K-1 | Dwarf.ctb | Layer: 1000/2000 | 50% Complete | 3:30 Remaining