luizribeiro/mariner #179: support for similar msla printers?

hak-isaiah, 2021-01-02:
I think it would be a cool feature to allow people to implement their own machine settings like the resolution and build area so we can use this with printers like the anycubic photon which lots of people dislike the fact that we cant wirelessly print with it and the process of using mariner is more efficient than using something like nanodlp

luizribeiro, 2021-01-02:
I think other printers that use chitubox should actually work with mariner. Maybe a few adjustments here and there would be required, but I think it should largely work.

However, I don't have access to anycubic photon or any other printers other than my Elegoo Mars Pro.

If anyone is willing to help here with testing, code or whatever I'm help to add official support for those printers.

hak-isaiah, 2021-01-02:
Oh great I'd be willing to help with testing and feedback i have a anycubic photon zero

luizribeiro, 2021-01-02:
@hak-isaiah, if you're willing to test with your anycubic photon zero, then give it a go!

I know that the folks from [photonsters](https://github.com/Photonsters/) have been able to get the Anycubic Photon with network on projects like this one: https://github.com/Photonsters/Universal-Photon-Network-Controller/

From looking at their code, the commands that they send to the printer are very similar (if not exactly the same), so I wouldn't be surprised if this worked as is. The instructions should be fairly similar to the ones I outlined [here](https://l9o.dev/posts/controlling-an-elegoo-mars-pro-remotely/).

That said, this is uncharted territory so proceed at your own risk :)

hak-isaiah, 2021-01-03:
@luizribeiro  the interface is great however the issue that stops this is i think the baud rates of the elegoo mars pro and the anycubic photon zero are different so the rpi  cant send commands to the photon yet and it cant read printer status and this doesn't accept .pw0 or .pws file types if there is anything i can do to help with this just message me

Shabbypenguin, 2021-01-03:
I get my phrozen sonic mini 4k on wednesday. Id love to be able to test this out and see what all works as is. looking through the code it seems its all just marlin commands. It would be incredible if this is the start of an octoprint like solution for SLA printers.

luizribeiro, 2021-01-03:
We should be able to add support for `.pw0` or `.pws` files. If you have some simple ones (like just a cube or whatever) I'm happy to look into adding support once I have more time. It shouldn't be too hard to do it if anyone is interested on giving it a shot.

About the baud rate, do you know what baud rate your printers are supposed to use? If so, you should be able to change it here:

https://github.com/luizribeiro/mariner/blob/1df5247e7ce5815ab29a57649e4b7ba40ee4b3fb/mariner/mars.py#L32

The support for `.pw0` or `.pws` files isn't strictly necessary to check if communication with the printer works, since the file format support is only used for previewing the files (so you'll likely not get any previews, but if you `scp` the `.pw0` or `.pws` file you may be able to get it to print).

Let me know what y'all find! 😄

hak-isaiah, 2021-01-04:
[Schizophrenia scan.zip](https://github.com/luizribeiro/mariner/files/5764212/Schizophrenia.scan.zip)
 it's a pw0 file of a brain scan that was turned into a stl

Shabbypenguin, 2021-01-09:
```
Traceback (most recent call last):
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/server/api.py", line 45, in print_status
    selected_file = elegoo_mars.get_selected_file()
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/mars.py", line 102, in get_selected_file
    self._extract_response_with_regex("ok '([^']+)'\r\n", data).group(1)
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/mars.py", line 39, in _extract_response_with_regex
    raise UnexpectedPrinterResponse(data)
mariner.exceptions.UnexpectedPrinterResponse
```

is the result when connecting to my sonic mini 4k. serial port not only was exposed but already had headers on it so connecting was a breeze. Just need to figure out mariner and i can have a remote UI/upload.

luizribeiro, 2021-01-09:
@Shabbypenguin, any chance this is because of #180 or do you consistently get that error when taking any action?

It's possible we need to make small tweaks to the protocol to support the Sonic Mini 4k, but want to first rule out that this isn't because of the existing issue.

Shabbypenguin, 2021-01-09:
> @Shabbypenguin, any chance this is because of #180 or do you consistently get that error when taking any action?
> 
> It's possible we need to make small tweaks to the protocol to support the Sonic Mini 4k, but want to first rule out that this isn't because of the existing issue.

Sadly I've never really delved into hardware hacking previously so this is a new experience for me. How did you go about reading from your serial printer in order to see what commands were used/needed? I see many of them are marlin used gcodes, but some like get selected file uses one I haven't seen before. I have a pi zero hooked up but I'm unsure of how to connect and start debugging to see if its simply a serial speed change, wrong commands, or something else.

for me mariner loads up and on the printer controls screen I get a spinning circle as it tries to load and connect before erroring out with that message. it is worth noting that i can upload ctb's and go to my printer and manually start them. I just cant do any of the core mariner functions/features.

luizribeiro, 2021-01-10:
@Shabbypenguin, nice. Glad you got the printer to reply to your serial commands. A few questions:

1. Which baud rate did you have to use?
2. Are the commands for the Sonic Mini 4K the same as the ones mariner is using? Which commands did you send for that output? 
3. Did any of mariner's features work?

Shabbypenguin, 2021-01-10:
1. 115200 like yours
2. im in the middle of a print, but so far I have tested get status, firmware version and z position (M4000, M4002, and M114 respectively) without issue or fail.
3. just upload and being able to select prints off the pi, that get selected file command loops and causes error after error on the UI.

My print will be going on overnight, so I wont be able to test and see the results of get selected file command until tomorrow.

Shabbypenguin, 2021-01-10:
Got it!

Select the file with M23. M24 tells it to start printing. Looked at the gcode commands from the photon controller app, looks like Phrozen stuck with the mostly standard ones as well. M27 lets me know its percentage done. So if that Ui issue from #180 can get resolved + make a more generic issue commands python with use whatever commands pulled from configs based on what type of printer you pick. I think this would probably get rolling and add at least 2 more printers support.

luizribeiro, 2021-01-10:
Can you summarize what are the differences between the Mars protocol and the Phrozen protocol?

You can tweak https://github.com/luizribeiro/mariner/blob/master/mariner/mars.py to make it work, put up a pull request once you confirm that it is working and I can refactor some of the code so we can have support for multiple printers.

I'm thinking of something along the lines of:

1. Creating an abstract class for all SLA printers, which `ElegooMars` will now inherit from
2. Setting up a separate class for Phrozen printers
3. Allowing users to pick which printer they are using from a config file

Shabbypenguin, 2021-01-10:
So far it just seems they use different gcode commands. phrozen opted for 

- M23 /filename.ctb
- M6030 /jeff.ctb ---> this line may not be needed, I have only tested it with this included currently
- M24

as the workflow to selecting a file and then printing it. Ill take some time to edit mars.py and hope it works.

Shabbypenguin, 2021-01-10:
[image]

it seems it needed no changes to mars.py. I happened to reboot everything on a clean boot and it didnt give me the get_selected_file() error as mentioned previously. So I was able to select the print via web ui and click print and it started. The ui updates in the background and shows progress. Every now and again I get the error from issue 180 but outside of that it looks like mariner already works well on phrozen printers.

luizribeiro, 2021-01-10:
Cool! I'll update the `README.md` to include the Phrozen Sonic Mini 4K. Thanks @Shabbypenguin!

@hak-isaiah, I'm going to keep this issue open for the Anycubic Photon. Let me know if I can help you with anything there.

crkives, 2021-09-30:
Hooked Up Anycubic Photon.  After working through a few issues.  The print status said ready.

Mariner doesn't take .photon files but I scp'd them to the directory and started a print from the printers ui.  At that point the UI gave 500 response.

Log:

> Sep 30 03:44:04 resin mariner[255]: AssertionError
> Sep 30 03:45:04 resin mariner[255]: ERROR:waitress:Exception while serving /api/print_status
> Sep 30 03:45:04 resin mariner[255]: Traceback (most recent call last):
> Sep 30 03:45:04 resin mariner[255]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/waitress/channel.py", line 397, in service
> Sep 30 03:45:04 resin mariner[255]:     task.service()
> Sep 30 03:45:04 resin mariner[255]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/waitress/task.py", line 168, in service
> Sep 30 03:45:04 resin mariner[255]:     self.execute()
> Sep 30 03:45:04 resin mariner[255]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/waitress/task.py", line 434, in execute
> Sep 30 03:45:04 resin mariner[255]:     app_iter = self.channel.server.application(environ, start_response)
> Sep 30 03:45:04 resin mariner[255]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/app.py", line 2464, in __call__
> Sep 30 03:45:04 resin mariner[255]:     return self.wsgi_app(environ, start_response)
> Sep 30 03:45:04 resin mariner[255]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/whitenoise/base.py", line 85, in __call__
> Sep 30 03:45:04 resin mariner[255]:     return self.application(environ, start_response)
> Sep 30 03:45:04 resin mariner[255]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/app.py", line 2450, in wsgi_app
> Sep 30 03:45:04 resin mariner[255]:     response = self.handle_exception(e)
> Sep 30 03:45:04 resin mariner[255]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/app.py", line 1867, in handle_exception
> Sep 30 03:45:04 resin mariner[255]:     reraise(exc_type, exc_value, tb)
> Sep 30 03:45:04 resin mariner[255]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
> Sep 30 03:45:04 resin mariner[255]:     raise value
> Sep 30 03:45:04 resin mariner[255]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
> Sep 30 03:45:04 resin mariner[255]:     response = self.full_dispatch_request()
> Sep 30 03:45:04 resin mariner[255]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
> Sep 30 03:45:04 resin mariner[255]:     rv = self.handle_user_exception(e)
> Sep 30 03:45:04 resin mariner[255]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception
> Sep 30 03:45:04 resin mariner[255]:     reraise(exc_type, exc_value, tb)
> Sep 30 03:45:04 resin mariner[255]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
> Sep 30 03:45:04 resin mariner[255]:     raise value
> Sep 30 03:45:04 resin mariner[255]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
> Sep 30 03:45:04 resin mariner[255]:     rv = self.dispatch_request()
> Sep 30 03:45:04 resin mariner[255]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
> Sep 30 03:45:04 resin mariner[255]:     return self.view_functions[rule.endpoint](**req.view_args)
> Sep 30 03:45:04 resin mariner[255]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/server/api.py", line 54, in print_status
> Sep 30 03:45:04 resin mariner[255]:     FILES_DIRECTORY / selected_file
> Sep 30 03:45:04 resin mariner[255]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask_caching/__init__.py", line 952, in decorated_function
> Sep 30 03:45:04 resin mariner[255]:     rv = f(*args, **kwargs)
> Sep 30 03:45:04 resin mariner[255]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/server/utils.py", line 19, in read_cached_sliced_model_fil
> Sep 30 03:45:04 resin mariner[255]:     file_format = get_file_format(filename)
> Sep 30 03:45:04 resin mariner[255]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/file_formats/utils.py", line 22, in get_file_format
> Sep 30 03:45:04 resin mariner[255]:     assert file_format is not None
> Sep 30 03:45:04 resin mariner[255]: AssertionError
> Sep 30 03:46:05 resin mariner[255]: ERROR:waitress:Exception while serving /api/print_status
> Sep 30 03:46:05 resin mariner[255]: Traceback (most recent call last):
> Sep 30 03:46:05 resin mariner[255]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/waitress/channel.py", line 397, in service
> Sep 30 03:46:05 resin mariner[255]:     task.service()
> Sep 30 03:46:05 resin mariner[255]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/waitress/task.py", line 168, in service
> Sep 30 03:46:05 resin mariner[255]:     self.execute()
> Sep 30 03:46:05 resin mariner[255]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/waitress/task.py", line 434, in execute
> Sep 30 03:46:05 resin mariner[255]:     app_iter = self.channel.server.application(environ, start_response)
> Sep 30 03:46:05 resin mariner[255]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/app.py", line 2464, in __call__
> Sep 30 03:46:05 resin mariner[255]:     return self.wsgi_app(environ, start_response)
[... cut ...]
> Sep 30 03:46:05 resin mariner[255]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/server/utils.py", line 19, in read_cached_sliced_model_fil
> Sep 30 03:46:05 resin mariner[255]:     file_format = get_file_format(filename)
> Sep 30 03:46:05 resin mariner[255]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/file_formats/utils.py", line 22, in get_file_format
> Sep 30 03:46:05 resin mariner[255]:     assert file_format is not None
> Sep 30 03:46:05 resin mariner[255]: 
>  AssertionError

sidvicius, 2022-05-31:
Anycubic Mono X  will be add to the list?

TheGreenJosip, 2026-02-11:
I'd also like to know that and maybe revive the thread. Also I am able to provide any information or dump since I own an AnyCubic Photon Mono X and would gladly contribute to add its support 👍🏼