luizribeiro/mariner #185: Elegoo Saturn

specsix, 2021-01-07:
I was going to try this out on the Saturn.  What are the chances the board is pretty much the same?

Also, do you think its worth my time to create a branch that allows mariner to control multiple Mars/Saturns from one mariner instance?

luizribeiro, 2021-01-08:
> I was going to try this out on the Saturn. What are the chances the board is pretty much the same?

Folks have been able to get mariner to run without any modifications on pretty much every Mars printer (original Mars, Mars Pro, Mars 2 and Mars 2 Pro), so I wouldn't be surprised that the Saturn uses the same mainboard.

If you do end up trying, please report it back so I can update the documentation (note however that I do not have access to a Saturn so I can't help debug much if the printer differs, I'm also not responsible for any damage of course :-)).

> Also, do you think its worth my time to create a branch that allows mariner to control multiple Mars/Saturns from one mariner instance?

That sounds like a neat idea. The tricky thing is that mariner runs on the Raspberry Pi that is directly connected to the printer.

On my setup I have a home server with nginx which runs as a reverse proxy. You could easily set up a similar reverse proxy that exposes all of your printers under the same host, with a simple `index.html` file that allows you to change between them. However, that wouldn't let you see status for all of your prints from the same page (which I imagine is a feature you're looking for).

One thing that we could potentially do to achieve that is:

1. Allow mariner to run in an API-only mode (i.e. no UI and no serving of static files from that node)
2. Have each printer run in one of such API-only nodes
3. Run a main instance of mariner, with a config file of all the IP addresses of the individual API-only mariner nodes. This would be the instance responsible for running the actual UI and allowing you to see all the printers somehow.

This would likely require a lot of changes to mariner, but I'm open to the idea if we're willing to be careful enough about the implementation and how to structure the code.

This does open a big can of worms which may require us to think about authentication between the nodes - but we don't support any sort of authentication and just assume the printer is behind a firewall or a reverse proxy with authentication, so maybe that's not a deal breaker for us to get this done. Maybe you can open a separate issue for this feature and we can take this discussion there?

specsix, 2021-01-08:
Thanks for the detailed response.   I definitely will return with my findings on the Saturn.  Going to try this soon.  

As far as mariner being able to view multiple printers, it sounds like it would be easier and cheaper (man hours wise) to just get another PI.   I probably will just run a local node server that views both mariner instances so they are combined on one page.   My larger goal is to have the Elegoo printer status show in Home Assistant.

luizribeiro, 2021-01-09:
Ah right, the solution I was proposing you would still need one Pi for each printer. There's no simple way around that.

Going to keep this open for us to keep track of support for the Elegoo Saturn.

rkolbi, 2021-02-05:
Confirmed working on Saturn.
https://ibb.co/zVPhGx3

specsix, 2021-02-05:
yup!  Got it working last week and forgot to post back.  Instructions were
a 1:1 match with the mars.   Also have it working in home assistant via
mqtt.

luizribeiro, 2021-02-05:
Thanks!

> Also have it working in home assistant via mqtt.

@specsix, can you elaborate on this? If you want to put up a tutorial on the wiki that would be awesome :)

rkolbi, 2021-02-05:
Another side note on the Saturn, no soldering is required as the pins are already installed on the board.

specsix, 2021-02-05:
My saturn required soldering.   Just like the build plates - not every
saturn is the same :(

commit b8c029cf (2021-02-06): Added Elegoo Saturn to list of supported printers

Fixes #185
closed 2021-02-06 by luizribeiro