luizribeiro/mariner #188: "Disk init fail" error wheh trying to print, Sonic Mini 4K

color65, 2021-01-11:
Hello, My setup is Raspberry Pi3, Raspbian OS, Mariner3D and Sonic Mini 4K. After uploading a ctb. file and try to print, the responce is "Disk init fail" Here are the full message from the Mariner3D;

Traceback (most recent call last):
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/server/api.py", line 208, in printer_command
    elegoo_mars.start_printing(filename)
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/mars.py", line 132, in start_printing
    self.select_file(filename)
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/mars.py", line 111, in select_file
    raise UnexpectedPrinterResponse(response)
mariner.exceptions.UnexpectedPrinterResponse: //Disk init fail!

luizribeiro, 2021-01-11:
Hey. This is a similar report to #152

It's most likely not a bug, but something wrong with parts of your USB Gadget setup. Can you read through that thread and see if any of the suggestions there are of any help?

color65, 2021-01-13:
Hello, after some research I think I found what is the problem. It seems that RPi 3 is not compatible for the task. Here is a quote from Raspberry forum:

**''A PI3 cannot act as an USB device, only as an USB master.

A zero can be an USB OTG device.

Main reason for other PI's why this isn't possible is because of the LAN9512 (or equivalent) chip in between USB ports and SoC."**

So I've just ordered a Pi Zero W and will test soon again. Thak you for the help!

luizribeiro, 2021-01-13:
Got it! Yeah, you need a Raspberry Pi that supports USB OTG.

I'll update `install.md` to reflect that and close this issue. Thanks @color65!