luizribeiro/mariner #199: API documentation?

NotExpectedYet, 2021-01-19:
Hey man, absolutely amazing of you to work on this I've been looking for an option for ages. 

Was wondering if there's any API documentation I am the developer of OctoFarm and adding support for this would be wonderful.

NotExpectedYet, 2021-01-19:
I have just found the API file. I will get this spun up and have a play.

luizribeiro, 2021-01-23:
Hey! Super excited that you're looking into this. Let me know if you need help or if any API changes will make things easier on your end.

I wasn't planning on having these API endpoints being used by things other than mariner's frontend itself, but I'm not opposed to it at all. I think this is a good idea :)

I think having documentation for the API at this point is probably overkill (all the endpoints are fairly simple and there's a bunch of test coverage for them), so let's rename this issue for OctoFarm support for now?

One thing that we may have to look into is if the printer status that we return is sufficient for the OctoFarm use case. Right now it's all pretty simple. Anyway, let me know!

NotExpectedYet, 2021-01-24:
> Hey! Super excited that you're looking into this. Let me know if you need help or if any API changes will make things easier on your end.
>
> I wasn't planning on having these API endpoints being used [...]

Awesome :D Good to have your backing. 

I agree totally with your regarding the documentations, I barely know Python and I managed to figure out the end points and formatting fine. 

OctoFarm uses two sets of printer status so I believe it should be fine with what you have. I categories things further myself in the system as I have to deal with the Hosts status and the Application as well. The only thing that would make this perfect but not required at all is a websocket connection. Currently that's how OctoPrint and Duet do it that I've seen and it saves on making a silly amount of API calls checking status for a print but it is no qualms to use API for now, 

https://github.com/NotExpectedYet/OctoFarm/discussions/225 - I have a discussion here for it as well, more for ideas on my end.

doucettom, 2021-03-05:
Just leaving a note here to follow-up as a flag to follow up !