luizribeiro/mariner #202: Creality LD-002H : first print completed, working

renatobo, 2021-01-20:
Completed thru the first print, no errors or issues so far with the exception of a few instances of `The printer returned an unexpected response: ''` as reported in #180 .

Creality LD-002H
v4.3.16_LCDKE / F2.21
Motheboard has a V3.2 print close the STM32 MCU

Install was fairly easy, from the side panel on the left:

- serial port pins exposed (in the picture below, between the WiFi header pin and the USB port)
- USB port on the board has an extension cable to relocate the external connector on the left side instead of behind: the cable can be removed (a bit of glue on it) so that no cable needs to run outside of the enclosure

Questions

- **power** I used the +5V pin in the serial connector and a GND pin nearby: raspberry pi does not complain
_do you know if this is too much load on the board?_
- **USB**: the 4 header pins next to it look very much the USB ones, this could be even better, 
_can anyone confirm?_

Board with serial header and USB (to be validated) header
[image]

The actual motherboard has more connectors than displayed in the manual

[image]

[image]

luizribeiro, 2021-01-23:
That's great! Thanks for reporting with such thorough pictures. I'll update the documentation to flag this printer as a supported model.

> power I used the +5V pin in the serial connector and a GND pin nearby: raspberry pi does not complain
do you know if this is too much load on the board?

No idea how much power the printer board can provide (and it may vary by printer model too?) nor how much power the Pi would need, so it's hard to say. Other people definitely have had success on this too, so you're definitely not alone.

If you want to, feel free to put up a pull request to update the install instructions to cover this possibility or anything specific to Creality printers.

closed 2021-01-27 by luizribeiro

LeonFedotov, 2021-01-29:
@renatobo hey! thanks so much for the report! i also have the same printer and got everything wired and working but now im wondering since the ui still is configured for elegomars so did you change any settings to adjust the size/exposure time if so then where ? :)

renatobo, 2021-01-29:
> @renatobo hey! thanks so much for the report! i also have the same printer and got everything wired and working but now im wondering since the ui still is configured for elegomars so did you change any settings to adjust the size/exposure time if so then where ? :)

Hi Leon, are you referring to the printer UI?
There is no change in firmware required for this (mariner) to work, are you referring to the firmware required by the WiFi dongle from chitubox instead?

luizribeiro, 2021-01-30:
> so did you change any settings to adjust the size/exposure time if so then where ? :)

Like @renatobo said, there's no changes required. All of the exposure and bed size settings are stored on the `.ctb` files created by your slicer, so mariner doesn't have much to do with them. Mariner just communicates with the printer to fetch print status, start prints, select which file should be printed, etc - it doesn't have to deal with any print parameters directly.

BalintBernhardt, 2021-01-30:
Sorry for asking it here. Maybe it's off topic.
@renatobo what pins you used to connected the RPi power? The VCC and GND pins (the ones next to the usb)? I ended up doing that after failed with the attempt to power from the 12V line on the PSU with a buck converter. I wanted to power the RPi all time independently from the printers mainboard. Sadly the GPIO GND wire is closing the circuit and turning on the printer not just the RPi.

LeonFedotov, 2021-01-30:
@renatobo, @luizribeiro thanks for the responses! you guys are totally right, the setup as is worked from the first try and no config adjustment was needed, i guess just that it says the wrong printer name confused me. ^_^

@BalintBernhardt me and @luizribeiro connected the pi power to the 5v next to the tx, rx pins, the pi will also power with an otg cable from either of its usb ports, unfortunately i think creality power switch turns off its internal 5v supply so you wont be able to source power for the pi from within with the  printer switched off.
pi powering the printer might be solved by like suggested cutting the 5v line on the usb cable or using diodes. hope this helps, cheers :)

luizribeiro, 2021-01-30:
> i guess just that it says the wrong printer name confused me.

If there's somewhere that the wrong printer name is being displayed on the latest version, please create an issue for it.

renatobo, 2021-01-31:
> @renatobo what pins you used to connected the RPi power? The VCC and GND pins (the ones next to the USB)?

**+5** -> as in the first picture, the one next to TX and RX
**GND** -> from the STM debugging connector (the one that's also highlighted in yellow with the writing "USB?") right behind the USB connector (the pin is named GND so it was hard to miss)

commit 3a82a41d (2021-01-27): Added Creality LD-002H as a supported printer

Fixes #202