luizribeiro/mariner #205: Raspberry pi serial connection to 5V?

boucz, 2021-01-21:
Hello
I did read that the raspberry pi gpio are 3V3 and not 5V tolerant. And the instructions talk about direct connections.
Seeing the 5V pin on the printer motherboard let me think about 5V signals.
Am I missing something? Or going wild with direct connections doesn't cause any problem?

Thanks a lot for your work!

color65, 2021-01-22:
Hello, which printer do you use? I have Sonic Mini 4k and for now it seems everything is ok. SM4K has a Chituboard, don't know if this is TTL or CMOS output for serial, but I think if there were 5V logical voltage, my Zero W would be damaged. Any idea how to measure the voltage of the serial output without oscilloscope?

boucz, 2021-01-22:
Hello,
I have a mars 2 pro, with with chitu L5.5 V2 series V3.1.7 motherboard
It is a good idea, I'll borrow an oscilloscope to check voltages levels :) my neighboor has one.

luizribeiro, 2021-01-23:
That's a good question :) I haven't plugged my scope on it, but I haven't had issues on my Elegoo Mars Pro and it seems like most folks haven't had any issues either. A bit silly/stupid of me of not checking it first, but I mostly went by the reports of people from [this thread](https://www.elegoomars.com/forum/showthread.php?tid=150) who had success at just plugging a Pi to the printer through those headers.

Maybe we just got lucky? Anyway, report back what you find please. We may want to adjust instructions and recommend folks to put some level converters or something.

boucz, 2021-01-25:
ok, the oscilloscope tells me 3.3V.
the chip on my motherboard is a STM32F401RCT6 witch is a 3.3V component (with all gpio tolerant to 5v).
No problem for me :) I can connect it directly to the raspberry pi.

My board is a  chitu L5.5 V2 series V3.1.7.

What are the microcontrollers on yours?

color65, 2021-01-26:
On my SM4K the microcontroller chip is STM32F407ZET6 - it seems the same like yours.

boucz, 2021-01-26:
> 
> 
> On my SM4K the microcontroller chip is STM32F407ZET6 - it seems the same like yours.

You're right, 3.3V io :)

luizribeiro, 2021-01-27:
Sounds good, going to close this then. Thanks for confirming!

closed 2021-01-27 by luizribeiro