luizribeiro/mariner #206: Raspberry pi camera streaming feature

dajosco, 2021-01-23:
First, what you've done here is awesome. I am definitely trying this next weekend.
I would suggest the possibility to add as a feature to stream video from the raspberry pi camera hooked up to its port. That would be really handy.

Thanks for your dedication!

luizribeiro, 2021-01-23:
Hey! I think this is a good idea, but I'd rather not have it as part of the core package - since not everyone has a camera.

I've been considering building a plugin system, which would allow people to build plugins and install them for features like this one.

What do you think?

BobLoblaw5, 2021-01-27:
I would also be very interested in being able to use a pi camera.  Would be really nice to be able to quickly check and make sure everything is still attached to the build plate.

Also thank you so much for making this program!  I just got it running on my Creality LD-002H (you should also be able to add the LD-002R to the list of compatible printers because the motherboards appear to be identical.)

Hypertoken, 2021-01-30:
I think a camera is a great idea as well. I've actually installed MJPG-streamer and view the camera in an iframe with mariner. I'd be great to have a bit more polish though. :)

However I have to say, as a side note; its very hard to get a good shot with the cover on my printer, and I just can't print with it off. ;) 
[image]

dogmachine, 2021-01-30:
> I think a camera is a great idea as well. I've actually installed MJPG-streamer and view the camera in an iframe with mariner. I'd be great to have a bit more polish though. :)
>
> However I have to [...]

i'm basically doing the same but with https://github.com/pikvm/ustreamer which is better suited for a rpi.  also will soon have x264 support.

anthonny-gunawan, 2021-02-03:
Waiting for this features, I put the Pi inside, have not tested what the camera see... but hopefully it will works

Soldering directly to make it smaller:
[image]

and mounting inside the cover:
[image]

dogmachine, 2021-02-04:
> Waiting for this features, I put the Pi inside, have not tested what the camera see... but hopefully it will works

I have been on a bit of a journey with the camera: https://imgur.com/a/iKbDWnW

DivineJimmi, 2021-02-05:
@Hypertoken Can you make a small tutorial how you implemented a camera stream into the webUI?

Hypertoken, 2021-02-05:
> @Hypertoken Can you make a small tutorial how you implemented a camera stream into the webUI?

@DivineJimmi. I wouldn't call it "Implemented" but this is what I did. 
1. Install Mariner; You probably already did this ;)
2. Install MJPG-streamer ( or ustreamer as @dogmachine suggests); there are plenty of tutorials out there for these.
3. Once the camera is working, I just wrote a simple HTML to display it, with Mariner in an iframe. 
[image]

DivineJimmi, 2021-02-05:
Tbh i have absolute no idea how to write a html file. Maybe you can provide me with a step by step guideline?

Hypertoken, 2021-02-05:
@DivineJimmi LOL, this really isn't the place for it, but... You can just use notepad to write the file. The screenshot I included above is the entire file.
>    [image]
[https://pastebin.com/pdfQLQig](url)
You just need to change the ip to your raspberry pi ip or hostname. Save the file with the html extension. Then open it in your browser.  You might need to adjust the top:500px tag to get the camera frame where you want it.

doucettom, 2021-03-13:
> @DivineJimmi LOL, this really isn't the place for it, but... You can just use notepad to write the file. The screenshot I included above is the entire file.
> 
> > [image]
> 
> https://pastebin.com/pdfQLQig
> You just need to change the ip to your raspberry pi ip or hostname. Save the file with the html extension. Then open it in your browser. You might need to adjust the top:500px tag to get the camera frame where you want it.

I'd need a little more guidance on this one. I've installed ustreamer and it nows streams to ip:8080/stream.

I created a camera.html file in the same directory that mariner store it's index.html. I open it with http://ip:5000/#/camera.html but it doesn't have the usual printer status nor it has the stream embedded but everything else of the mariner dashboard is there

daxliniere, 2021-05-15:
All of this and more can be done with Node-RED which comes pre-installed on Raspberry Pi OS (including Light). I am actually in the process of building a platform to do all of this and more.
Definitely check out Node-RED, it's free, awesome and the community is very nice and helpful.

All the best!
Dax.
[image]

DivineJimmi, 2021-05-15:
This looks really good. So for my understanding in the future this is easy to implement? I have absolutely no idea how to do this with nodenred.

daxliniere, 2021-05-15:
@DivineJimmi, I didn't know anything about Node-RED until 2 months ago. :)
Though I am an engineer and can think logically, I'm definitely not a programmer, so I'm sure you can learn NR, too.

Once I've got my system all figured out, I'm going to post the entire flow to import directly into NR. Then you just need to select which sensors and output relays you have chosen to connect.

DivineJimmi, 2021-05-15:
@daxliniere So you are connecting your sensors to the GPIO pins and choose them in your setup? Is this working with a camera too?

daxliniere, 2021-05-15:
@DivineJimmi, yes exactly. NR interfaces with all manner of things using modules, such as the Pi modules that come preinstalled.
I have not got this particular setup working with a camera, but I have another Pi which takes a feed from a network (IP) camera and emails it to me upon a specific set of conditions, and the Pi-Zero has a camera connector on-board so I'm sure it would be a piece of cake to get the camera working, too.

daxliniere, 2021-05-15:
@DivineJimmi, just find the Node-RED Youtube channel and watch their tutorials playlist. That'll give you an idea of how simple, but powerful it is.

Kaympe20, 2024-06-29:
> I think a camera is a great idea as well. I've actually installed MJPG-streamer and view the camera in an iframe with mariner. I'd be great to have a bit more polish though. :)
>
> However I have to [...]

How did you do an iframe on the web page? Did you just edit the underlying html?