luizribeiro/mariner #218: Creality LD-002R working

BalintBernhardt, 2021-01-29:
Huge thanks @luizribeiro!
Working nicely with the R version also.

Creality LD-002R
Firmware: v4.3.1_LCDK / F2.18
Motheboard has V3 printed next to UART

luizribeiro, 2021-01-29:
Thanks! I'll update the `README.md` with that info soon.

closed 2021-01-30 by luizribeiro

anthonny-gunawan, 2021-02-01:
Hello @BalintBernhardt , did the correct Printer name show up and you can print via UI?

BalintBernhardt, 2021-02-02:
@anthonny-gunawan I saw #220. For me nothing has been displayed under Printer Status. Printing from UI works nicely except the issue described in #180.

anthonny-gunawan, 2021-02-04:
I am still unable to print via UI, it give me this error:
[image]

the Details:
Traceback (most recent call last):
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/server/api.py", line 208, in printer_command
    elegoo_mars.start_printing(filename)
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/mars.py", line 140, in start_printing
    raise UnexpectedPrinterResponse(response)
mariner.exceptions.UnexpectedPrinterResponse: Error:Unknown format, file /Creality_LD-002R_Feets_2.ctb

luizribeiro, 2021-02-04:
Are you running the same firmware as @BalintBernhardt?

> Firmware: v4.3.1_LCDK / F2.18

Most likely your printer's firmware is older than that and doesn't support `.ctb` files?

anthonny-gunawan, 2021-02-05:
Let me check the Firmware, but I can print just Print (for the same file) via the Touchscreen interface

commit c419dce8 (2021-01-30): Add Creality LD-002R to list of supported printers

Fixes #218