luizribeiro/mariner #219: HTTP Error Codes?

samurad27, 2021-01-30:
I'm not very savvy and am embarrassed to have to ask this but I'm totally at a loss... can anyone help me figure out why the mariner app in the browser is giving an HTTP 400 error code when I attempt to upload a file and an HTTP 500 error code when I go to the home screen?

closed 2021-01-30 by samurad27

xrectifyx, 2021-04-17:
@samurad27 what did you end up doing to resolve your issue? im stuck on a regular mars with the same issue and traceback.

samurad27, 2021-04-17:
Idk if this will help, but the night after I wrote that question I had an epiphany at around 3 am, schlepped down to my workshop in my PJ’s and just plugged the mini-usb adapter into the other port on the Pi, after that it worked like a charm!