luizribeiro/mariner #220: Creality LD-0002R is identified as Elegoo Mars Pro

anthonny-gunawan, 2021-02-01:
I have installed this successfully, but my printer is identified as Elegoo Mars Pro (it is installed to Creality LD-0002R - how do I change this to show the correct Name?

This also made me cannot print from the UI as it said that the format .ctb are not recongnized

Thank you,
Anthonny

anthonny-gunawan, 2021-02-01:
I can print using the LCD, and can see the progress, but sometime it gave me this error (I think it is due to it recognized the Printer incorrectly):

Traceback (most recent call last):
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/server/api.py", line 44, in print_status
    selected_file = elegoo_mars.get_selected_file()
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/mars.py", line 102, in get_selected_file
    self._extract_response_with_regex("ok '([^']+)'\r\n", data).group(1)
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/mars.py", line 39, in _extract_response_with_regex
    raise UnexpectedPrinterResponse(data)
mariner.exceptions.UnexpectedPrinterResponse

luizribeiro, 2021-02-02:
To be clear here, mariner doesn't really attempt to recognize the printer at all.

Where are you seeing that your printer is identified as an Elegoo Mars Pro?

> I can print using the LCD, and can see the progress, but sometime it gave me this error (I think it is due to it recognized the Printer incorrectly):

This is actually due to #180. It's a known transient issue. It doesn't really affect anything. It'll get fixed once I have a bit more of time to dedicate to this or someone puts up a pull request.

anthonny-gunawan, 2021-02-02:
[image]
It is actually what I see in the home page - I thought that there are some miscommunications due to Printer Type

BobLoblaw5, 2021-02-02:
Are you sure you don't have the printer profile set to Mars Pro in Chitubox?  It's possible it's pulling that from the .ctb file.

anthonny-gunawan, 2021-02-02:
@BobLoblaw5 , It was identified as Elegoo Mars Pro even without .ctb files

luizribeiro, 2021-02-02:
@anthonny-gunawan, you're using an outdated version of the `.deb` package.

This was a hard-coded string from when we only supported the Elegoo Mars Pro. It was removed on 77838c3438078b025f050a9efd453b306848c188, which was part of the v0.1.1-1 release.

Closing this since it's not really an issue.

closed 2021-02-02 by luizribeiro

anthonny-gunawan, 2021-02-02:
Aha, how do I update the version?

luizribeiro, 2021-02-02:
Download the latest package from the [releases page](https://github.com/luizribeiro/mariner/releases/) and install it with `dpkg -i`.

Maybe what put you off is that [these instructions](https://github.com/luizribeiro/mariner/blob/master/docs/install.md) have the link to v0.1.0 and not v0.1.1?

Maybe there's a way to make that always link to the latest version....