luizribeiro/mariner #235: HTTP 500 status code / SCP no permissions / not seeing files on printer

DivineJimmi, 2021-02-04:
Hey, i have those following 3 problems:

1. Connecting to <raspberrypi-ip>:5000 works flawless. Clicking on Files responds with an "The server replied with a 500 HTTP status code." error. I cannot upload anything.

2. In a previous installations i was able to upload, but i havent seen any files on my printer menu. (Didnt even try to print from webUI)

3. I cannot upload Files to the Folder mnt/usb_share. With WinSCP it says, i have no permissions. Even if i try to mount it as a network share on windows, i can browse and see all the files, but i have no permissions to write into that folder/share. Both times i am using the standard "pi" user with "raspberry" as password.

DivineJimmi, 2021-02-04:
Issue 1: After uploading a file via webUI the 500 HTTP error message disappeared.
Issue 2: i can now see the uploaded files on my printer. I made some changes in the rc.local. #152 helped a lot.
Issue 3: Still an issue. No way to upload via SCP.
Issue 4: The printer prints now. But with following error response: "The printer returned an unexpected response: 'ok N:0\r\n'"

DivineJimmi, 2021-02-04:
Wow. i cannot believer this. After starting a print like 5 times. The errors always changed and now they disapperaed. Unfortunately i am still having issues with Issue 3.

luizribeiro, 2021-02-05:
> I cannot upload Files to the Folder mnt/usb_share. With WinSCP it says, i have no permissions. Even if i try to mount it as a network share on windows, i can browse and see all the files, but i have [...]

Have you tried adding your `pi` user to the `mariner` group?

If you look at the install instructions, specifically at the `fstab` file, we generally mount `/mnt/usb_share` as owned by the `mariner` group of users, which has read/write access to all the files under that mount point.

This means you won't be able to write to `/mnt/usb_share` unless the user you are writing as is part of the `mariner` group.

DivineJimmi, 2021-02-05:
That worked perfectly. For anyone having the same issues: SSH to your Raspberry and type following: `sudo adduser pi mariner`

closed 2021-02-05 by DivineJimmi