luizribeiro/mariner #243: New Feature

DivineJimmi, 2021-02-05:
Webcam feature was already requested. But what about a timelapse feature? So your timelapse starts, when the print start and ends with the print end. The finished files should be accessible through the webUI.

Would be a really nice feature.

DivineJimmi, 2021-02-07:
Just had another idea for a timelapse feature. If there is an auto-timelapse mode implemented, which is making a video, every time a print starts, it would be nice to make a "minimum layer" option. Because you wouldnt see anything on a timelapse with a print that has only 100 layers.

bludin, 2021-05-26:
I would like to extend on the idea: while below the minimum layer, it would be great to have an option to pause the print every n-th layer, drive the build plate up to a defined position, take a pic, lower it again and resume the print. That would slow down the print a bit but allow the user to see the early, critical layers forming.

ixtenebrae, 2021-05-26:
I've seen people do this with a light sensor placed in the chassis,
connected to the trigger on a Canon DSLR. The moment the LEDs turn on seems
to be a great moment to capture an image. But this also means smaller items
(like the keycaps I'm printing now) are going to be worthless. I was
thinking if Mariner sent a periodic M25 and took the picture while paused,
you could get some good timelapses. It'd add some extra time to the print
though.

bludin, 2021-05-27:
It would indeed, but it would also allow you to abort the print early on when things go wrong, which will save you a lot of time