luizribeiro/mariner #246: Basic plugin system

luizribeiro, 2021-02-07:
There's a lot of interesting feature requests that require additional hardware to have setup (temperature sensors, tracking with cameras, etc).

Because of the additional hardware requirements, I don't think it makes sense to have those features in the core mariner code. So we probably need a plugin system of sorts that allows these features to be installed.

xmodpt, 2021-03-24:
i have to agree....
Some of the thins an SLA printer requiers is a heater controled via a thermostat.
One good thing could be using a basic sensor ou a DHT11 that reads Temp and Humidity to control a relay to turn on and off a heater then a Max and Min value are reached.

Another could be a camera (pi cam as a cheap option) to control visualy if the print has faild and that way STOP the print.

I am not much of  a programer but i have all the hardware, so i will try to get "something to work" and see where it leads us.

Up so far... GOD WORK ... i am loving it ....

Kind Regards / Obrigado
Nuno Pires

hansale, 2021-03-25:
I'm a programmer but don't really know much about the hardware. I'd like to help, we need to come up with a uniform interface for plugins.
Off the top of my head the things we need to hammer out are

- front end we'd want some sort of paneling/homepage that displays a list of installed plugins, and you can enable, disable them.
- possibly the ability to change ordering of how the plugins are displayed on homepage
- define an extensible API served by the platform, maybe keep /api/plugins/* reserved for plugins? Doing a GET on /api/plugins  could poll and return a list of the plugins themselves, with information of what data is consumed? So if I have a /api/plugins/tempcheck plugin I do a GET and it returns the current temp info.

xmodpt, 2021-03-25:
In the hardware... i can help

so far i have a small Python script that read temp and humidity and prints is on screen
Temperatura = 24.0 C Humidade = 94.0%
Aguarda 2 segundos para efetuar nova leitura...n
Temperatura = 23.0 C Humidade = 94.0%
Aguarda 2 segundos para efetuar nova leitura...n
Temperatura = 23.0 C Humidade = 94.0%
Aguarda 2 segundos para efetuar nova leitura...n
Temperatura = 23.0 C Humidade = 94.0%
Aguarda 2 segundos para efetuar nova leitura...n
Temperatura = 23.0 C Humidade = 94.0%
Aguarda 2 segundos para efetuar nova leitura...n
Temperatura = 23.0 C Humidade = 94.0%

and i have hardcoded MAX limit and MIN limit to activate a relay 
in other words, when printer is truned on, the temp is read and in it is under MAX value (30ºC) it activates the relay to trun on the heater.´
When MAX temp is reached, the relay is turn off (turning the heater OFF) and when it reaches MIN Temp (24ºC) it turns on the relay (heater) to start the heating process again.... and it keeps looping.

What i can't do is to convert that info (temp ºC) to a widget and i don´t know how to turn the hard coded values (Max Temp and Min Temp) in to variables that can be changed by the user.

Sample Code:
`
import Adafruit_DHT
import RPi.GPIO as GPIO
import time

sensor = Adafruit_DHT.DHT11 

GPIO.setmode(GPIO.BOARD) # Activate GPIO Pins
GPIO.setup(26, GPIO.OUT) # Connect Relay Data pin to GPIO 26

print("Reading Temp and Humidity values :");

while(1): 
	humid, temp = Adafruit_DHT.read_retry(11, 17);
	if humid is not None and temp is not None:
		print ("Temperature = {0:0.1f} C Humidity = {1:0.1f}%").format(temp, humid);
		print ("Wating 2 sec until next reading");
		time.sleep(2) # Sleep 2 sec before next reading
		if temp>= 30: # MAX TEMP SET TO 30ºC

				GPIO.output(26, 0) # Relay pin set to "0" (OFF)

		else:
				GPIO.output(26, 1) # Relay pin set to "1" (ON)

		if temp<=25: # MIN TEMP SET TO 25ºC
				GPIO.output(26, 1) # Relay pin set to "1" (ON)
		else:
				GPIO.output(26, 0) # Relay pin set to "0" (OFF)
     	# Mensagem de erro de comunicacao com o sensor
     	#		print("Falha ao ler dados do DHT11 !!!")
`

Hope this helps

Nuno Pires

xmodpt, 2021-03-25:
for some reason cant past the script as code :(

luizribeiro, 2021-03-28:
> I'm a programmer but don't really know much about the hardware. I'd like to help, we need to come up with a uniform interface for plugins.
> Off the top of my head the things we need to hammer out [...]

@hansale, this overall seems reasonable. I think I would just skip the frontend for managing plugins and UI ordering for now and focus on how the plugins would be implemented and how they would talk to mariner. We could keep things simple and just [auto-discover plugins](https://packaging.python.org/guides/creating-and-discovering-plugins/) that are on a specific path like `/opt/mariner/plugins` or something. Using [`flask-plugins`](https://flask-plugins.readthedocs.io/en/master/) might be interesting too, but I haven't looked enough into it.

Plugins would have to be able to do these on the backend:

1. Declare which major version of mariner is supported (i.e. every time we make a breaking change on the API for plugins, we would bump the major version). We would validate this upon plugin discovery and ignore plugins that don't support the major version that is running.
2. Add new flask routes. I like the idea of keeping all of them under `/api/plugins/<plugin-name>/` to prevent conflicts.
3. Add static resources to be served by mariner (so we can serve JS, etc)

On the frontend, plugins would have to be able to:

1. Register new cards for the homepage
2. Register new views
3. Register new entries on the drawer menu

Does this seem reasonable? Anything else that we're missing? Once we know everything that plugins need to do we can design what the plugin system could look like.

hansale, 2021-03-29:
@xmodpt I saw an ingenious time lapse that uses a light sensor to tell if the resin printer is on, and the camera snaps a picture right as the UV light turns on, allowing for perfect time lapses regardless of the cure times of the device. Very simple and cool.

Are you using a temperature sensor like this one? https://www.adafruit.com/product/381 

@luizribeiro As for the plugin system, I was just spitballing "ideal world" scenarios with the drag and drop, obviously a drag and drop UI for plugin ordering would be orders of magnitude more complex than having , especially considering that by virtue of running Mariner at all I'd guess the average technical proficiency of the users is substantially higher than the average.

This sounds like a good start, although it seems like #75 would need to be implemented before substantial headway could be made? As we'd want plugins to have the ability to access the printer, right?

luizribeiro, 2021-03-29:
> This sounds like a good start, although it seems like #75 would need to be implemented before substantial headway could be made? As we'd want plugins to have the ability to access the printer, right?

I'm not sure if we need #75 done before we can get started on the plugin system.

Maybe for us to have a next release with the plugin system we would want both issues closed, but we can probably get the architecture of the plugin system in place without #75

xmodpt, 2021-03-29:
@hansale 
Hello m8 

yes that one is the target one, however i am testing with a cheap DHT11 module [this one DHT11](https://pt.aliexpress.com/item/32840892862.html?src=google&albch=shopping&acnt=708-803-3821&isdl=y&slnk=&plac=&mtctp=&albbt=Google_7_shopping&aff_platform=google&aff_short_key=UneMJZVf&&albagn=888888&isSmbAutoCall=false&needSmbHouyi=false&albcp=9437679086&albag=97536833404&trgt=296904914040&crea=pt32840892862&netw=u&device=c&albpg=296904914040&albpd=pt32840892862&gclid=Cj0KCQjw9YWDBhDyARIsADt6sGZZP5QonLmYOQhaqstM7_1t13zE29oldHZQuTKKszONLyPsjAgNDCgaAsjNEALw_wcB&gclsrc=aw.ds)

daxliniere, 2021-05-15:
All of this and more can be done with Node-RED which comes pre-installed on Raspberry Pi OS (including Light). I am actually in the process of building a platform to do all of this and more.
Definitely check out Node-RED, it's free, awesome and the community is very nice and helpful.

All the best!
Dax.

luizribeiro, 2021-05-15:
Hey folks, let's keep this issue on the topic of the plugin system. If you'd like to discuss something that is unrelated to development of `mariner`, please use [Discussions](https://github.com/luizribeiro/mariner/discussions) instead of Issues.