luizribeiro/mariner #260: Print status stuck on first layer

zjpiazza, 2021-02-15:
I was successfully able to connect Mariner with my Creality LD002R and start a print via the web interface. I went over and looked at the printer display and saw that it appeared to have accepted the print job and showed the status being on layer one of 1200 or something. However, the build plate did not descend into the resin chamber and it just stayed stuck there with no signs of activity. Any ideas what may be going on here?

Thanks!

luizribeiro, 2021-02-15:
That's weird. Can you clarify a bit what you mean by "accepted the print job"? What does the printer display show? What does mariner show?

Maybe you can try connecting `minicom` to `/dev/serial0` while you send the commands from `mariner` and see what the printer is returning to each command?

@BalintBernhardt may have ran into this before, as he reported the printer as working on #218. I don't have a Creality printer myself..

zjpiazza, 2021-02-15:
Like when I look at the LCD I can see the job percentage as 1% but nothing else is happening

zjpiazza, 2021-02-15:
Also I'm curious why you hardcode the class to be named  ElegooMars when it's supposed to work for a variety of printers?

luizribeiro, 2021-02-15:
> Like when I look at the LCD I can see the job percentage as 1% but nothing else is happening

I think we'll need more information here to debug this. It would be good if you could:

1. Do the `minicom` steps I mentioned above and see what's going on on the communication between the printer and the pi
2. Mention which firmware you are using
3. Check if printing the exact same file from the printer itself works and if it's just from the mariner UI that the printing job is getting stuck
4. Mention which file you're trying to print

> Also I'm curious why you hardcode the class to be named ElegooMars when it's supposed to work for a variety of printers?

It's the printer I have and the first printer that we got working. This can be renamed nowadays and has no effect. Let's keep this thread on-topic though. I'm happy to take a look at a PR if you want to rename things

luizribeiro, 2021-04-16:
I believe this is the same issue as described on #311. Since this issue has gotten pretty stale, I'm going to close it. If the fix for #311 doesn't end up solving it once we have a new release with the fix, feel free to reopen this issue.

closed 2021-04-16 by luizribeiro