luizribeiro/mariner #296: Yet another HTTP 500

cbanderson22, 2021-03-05:
Hey there, running into a bit of a snag getting this up and running on my Saturn. My printer can see files on the Pi fine, but the homepage just gives me the 500 error. Here is what I get in the log:
[image]

I'm terribly new to any of this, is there something I've overlooked? Thanks!

cbanderson22, 2021-03-06:
Alrighty, got it figured out.

closed 2021-03-06 by cbanderson22

luizribeiro, 2021-03-06:
@cbanderson22, you may want to explain what the problem was so others who run into it can use your solution. What was wrong here?

doucettom, 2021-03-07:
I have pretty much the same issue, I'm interrested in how it was solved as well :)