luizribeiro/mariner #301: Voxelab Proxima

doucettom, 2021-03-05:
I'll get it in a few days from now. It is compatible with Chitubox but I don't know anything else. I just hope that the pins are already on the board

luizribeiro, 2021-03-06:
Sounds good, I'll leave this assigned to you then. Thanks for testing another printer!

doucettom, 2021-03-06:
I opened the side case and seems like the headers are already there !

doucettom, 2021-03-07:
Is there any way to directly communicate to sort some issues out @luizribeiro

doucettom, 2021-03-07:
Mar  6 20:48:38 raspberrypi mariner[609]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/mars.py", line 171, in _send_and_read
Mar  6 20:48:38 raspberrypi mariner[609]:     response = self._serial_port.readline().decode("utf-8")
Mar  6 20:48:38 raspberrypi mariner[609]: UnicodeDecodeError: 'utf-8' codec can't decode byte 0xb4 in position 5: invalid start byte
Seems like I have the same serial problems than some other users.

config.txt:
pi@raspberrypi:~ $ cat /boot/config.txt 
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
#dtoverlay=vc4-fkms-v3d
#max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d
dtoverlay=dwc2
enable_uart=1

cmdline.txt
pi@raspberrypi:~ $ cat /boot/cmdline.txt 
console=tty1 root=PARTUUID=4640d5e8-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait modules-load=dwc2

I have an error every time I upload a file, even though it seems to upload.

doucettom, 2021-03-07:
Linux raspberrypi 5.10.17+ #1403 Mon Feb 22 11:26:13 GMT 2021 armv6l

doucettom, 2021-03-07:
OK I think I kinda got it partially working
ok
ok '6��-����ģ�� 20-2.5-20200714.fdg'
ok '6��-����ģ�� 20-2.5-20200714.fdg'
//Unknown opt: m  0x6d index:0 len:1,line:0,s:m
ok
//Unknown opt: 1  0x31 index:0 len:1,line:0,s:1
//Unknown opt: 1  0x31 index:0 len:1,line:0,s:1
//Unknown opt: 5  0x35 index:0 len:1,line:0,s:5

I receive regular ok's from printer.

doucettom, 2021-03-08:
OK, so I started to investigate a tad deeper. I ruled out serial port from printer and/or raspberry Pi. I can send gcode if I connect via minicom and send Gcodes, I get the proper reply from the printer so this means not all hope is lost !

Mar 07 23:50:13 raspberrypi mariner[724]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/mars.py", line 171, in _send_and_read
Mar 07 23:50:13 raspberrypi mariner[724]:     response = self._serial_port.readline().decode("utf-8")
Mar 07 23:50:13 raspberrypi mariner[724]: UnicodeDecodeError: 'utf-8' codec can't decode byte 0xb4 in position 5: invalid start byte
Mar 07 23:50:13 raspberrypi mariner[724]: ERROR:waitress:Exception while serving /api/print_status
Mar 07 23:50:13 raspberrypi mariner[724]: Traceback (most recent call last):
Mar 07 23:50:13 raspberrypi mariner[724]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/waitress/channel.py", line 350, in service
Mar 07 23:50:13 raspberrypi mariner[724]:     task.service()
Mar 07 23:50:13 raspberrypi mariner[724]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/waitress/task.py", line 171, in service
Mar 07 23:50:13 raspberrypi mariner[724]:     self.execute()
Mar 07 23:50:13 raspberrypi mariner[724]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/waitress/task.py", line 441, in execute
Mar 07 23:50:13 raspberrypi mariner[724]:     app_iter = self.channel.server.application(environ, start_response)
Mar 07 23:50:13 raspberrypi mariner[724]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/app.py", line 2464, in __call__
Mar 07 23:50:13 raspberrypi mariner[724]:     return self.wsgi_app(environ, start_response)
Mar 07 23:50:13 raspberrypi mariner[724]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/whitenoise/base.py", line 85, in __call__
Mar 07 23:50:13 raspberrypi mariner[724]:     return self.application(environ, start_response)
Mar 07 23:50:13 raspberrypi mariner[724]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/app.py", line 2450, in wsgi_app
Mar 07 23:50:13 raspberrypi mariner[724]:     response = self.handle_exception(e)
Mar 07 23:50:13 raspberrypi mariner[724]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/app.py", line 1867, in handle_exception
Mar 07 23:50:13 raspberrypi mariner[724]:     reraise(exc_type, exc_value, tb)
Mar 07 23:50:13 raspberrypi mariner[724]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
Mar 07 23:50:13 raspberrypi mariner[724]:     raise value
Mar 07 23:50:13 raspberrypi mariner[724]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
Mar 07 23:50:13 raspberrypi mariner[724]:     response = self.full_dispatch_request()
Mar 07 23:50:13 raspberrypi mariner[724]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
Mar 07 23:50:13 raspberrypi mariner[724]:     rv = self.handle_user_exception(e)
Mar 07 23:50:13 raspberrypi mariner[724]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception
Mar 07 23:50:13 raspberrypi mariner[724]:     reraise(exc_type, exc_value, tb)
Mar 07 23:50:13 raspberrypi mariner[724]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
Mar 07 23:50:13 raspberrypi mariner[724]:     raise value
Mar 07 23:50:13 raspberrypi mariner[724]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
Mar 07 23:50:13 raspberrypi mariner[724]:     rv = self.dispatch_request()
Mar 07 23:50:13 raspberrypi mariner[724]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
Mar 07 23:50:13 raspberrypi mariner[724]:     return self.view_functions[rule.endpoint](**req.view_args)
Mar 07 23:50:13 raspberrypi mariner[724]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/server/api.py", line 45, in print_status
Mar 07 23:50:13 raspberrypi mariner[724]:     selected_file = elegoo_mars.get_selected_file()
Mar 07 23:50:13 raspberrypi mariner[724]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/mars.py", line 100, in get_selected_file
Mar 07 23:50:13 raspberrypi mariner[724]:     data = self._send_and_read(b"M4006")
Mar 07 23:50:13 raspberrypi mariner[724]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/mars.py", line 171, in _send_and_read
Mar 07 23:50:13 raspberrypi mariner[724]:     response = self._serial_port.readline().decode("utf-8")
Mar 07 23:50:13 raspberrypi mariner[724]: UnicodeDecodeError: 'utf-8' codec can't decode byte 0xb4 in position 5: invalid start byte
Also, file upload stop generating errors as long as I'm uploading "legit" ctb files. Just renaming a .txt to .ctb was bugging. I highy suggest a validation of file types as some folks might want to upload something else than .ctb or ctbdlp. Now, that being said, I will need some help @luizribeiro with the serial errors I currently get.

doucettom, 2021-03-11:
Got a little time to work on it tonight. Here's how I pretty much made it work. I had to update the Chitu firmware on the printer to 4.3.9 and now everything works fine.I only updated FPGA and Core functions, everything else stayed as it was ! I will test print in the next few days.

doucettom, 2021-03-14:
OK I think I found out why it wasn't working with original firmware. Voxelab are using fdg files instead of .ctb.

luizribeiro, 2021-03-15:
@doucettom, so I guess it's confirmed that mariner works with the Voxelab Proxima, but the printer has to be updated with the most recent firmware?

doucettom, 2021-03-15:
It will but flashing it with a regular Chitu firmware is effin things up with preconfigured slicer profile as they all output .fdg files but the printer won't print them anymore and mariner doesnt support them. Reserve is true with stock firmware as it won't print ctb. I opened an issue for .fdg support.

luizribeiro, 2021-03-15:
Cool, thanks for looking into this! Let's keep this issue open until #312 is closed then, after which we can cut a new release and update the `README.md` file with the new printer.

doucettom, 2021-03-19:
I reverted the printer to stock firmware. My mariner 3d instance is still connected to the printer. Let me know if you want me to test something.

doucettom, 2021-03-30:
Maybe this could help a bit. I will try to make something but last time I did dev stuff was almost 20 years ago:
https://github.com/ezrec/uv3dp

luizribeiro, 2021-04-11:
@doucettom, I recently added support for `.fdg` files. Can you try installing the `.deb` from this nightly build and see if it works? https://github.com/luizribeiro/mariner/actions/runs/739255134

The version says 0.1.1, but it's the `.deb` associated with that commit, so it should work with `.fdg` files.

I can't really test much other than just checking that the `.fdg` file is parsed correctly, since I don't have a Voxelab printer available.

doucettom, 2021-04-11:
I'll test this asap. I have two Voxelab Proxima :)

doucettom, 2021-04-16:
Allright ! did some testing today. So far I cannot upload .fdg files from the web ui, it results in an http 400 error. If I upload it through scp, it shows but I got a message that this file cannot be printed.

luizribeiro, 2021-04-17:
@doucettom, ops. My bad. I should have tested uploading it myself first at least. I forgot to update some files that have a hard-coded list of supported files. There's some cleanup we should do now that we support 3 different file formats :)

Can you try the artifact generated by this CI run? https://github.com/luizribeiro/mariner/actions/runs/757581396

doucettom, 2021-04-17:
I printed the test file I uploaded above. Should be done in a few minutes. Preview is working as well ! Awesome :) Let me write something up Proxima-specific. I'll also post in Voxelab Groups (Official and unofficial :+1: )

luizribeiro, 2021-04-17:
Cool! I'll update the list of supported printers on the `README` file and close this issue then. I'll try to cut a new release this weekend that will include support for the Voxelab Proxima. But I want to get #231 done first, so folks can more easily upgrade mariner whenever there's a new version.

Thanks for testing this, @doucettom!

closed 2021-04-17 by luizribeiro

luizribeiro, 2021-04-17:
Just FYI that v0.2.0 is out with support for the Voxelab Proxima in. See announcement at https://github.com/luizribeiro/mariner/discussions/377

commit e85c56ed (2021-04-17): Added Voxelab Proxima to list of supported printers

Fixes #301