luizribeiro/mariner #302: http500 on photon mono se

EXilE-EXistance, 2021-03-05:
trying to setup at least some functionality on a mono se, these error could be unrelated
[image]
[image]

luizribeiro, 2021-03-06:
This seems like a problem with how you setup the serial port, probably similar to what was reported on #296.

Did you follow the install instructions? What does your `config.txt` look like?

cbanderson22, 2021-03-07:
Yeah, looks same as my log. I got mine fixed after starting from scratch and literally copy and pasting the commands. 3rd time was the charm.

doucettom, 2021-03-07:
> Yeah, looks same as my log. I got mine fixed after starting from scratch and literally copy and pasting the commands. 3rd time was the charm.

You didn't found what was the culprit ?

cbanderson22, 2021-03-07:
Probably just inputted something wrong the first couple of times. I did not know you can copy text and right click in the console to paste. Got rid of the http 500 after doing it like that.

doucettom, 2021-03-07:
> Probably just inputted something wrong the first couple of times. I did not know you can copy text and right click in the console to paste. Got rid of the http 500 after doing it like that.

I still got some problems get the serial port going. Mine isn't a photon though.

ShlongDongSilver, 2021-03-08:
yeah very well could be a bad config of mine, mostly posted due to it being a photon.
[image]
I did follow the install instructions, I though it might have been something with the "dtoverlay=dwc2" being added after the original dtoverlay, but removing the first one made no change.
uart is also enabled.

doucettom, 2021-03-11:
I think you need to add it to section [all] instead of pi4

luizribeiro, 2021-03-29:
Seems like this has been solved with the comment above, so closing the issue.

closed 2021-03-29 by luizribeiro

jhbruhn, 2021-12-09:
@ShlongDongSilver did you have success connecting to a serial port on the Photon Mono SE? I have a Photon Mono but I'm struggling to find the correct pinout. As it seems, the motherboards for those two printers are the same.

MauDib420, 2021-12-14:
>  the motherboards for those two printers are the same.

Unfortunately, the Photon Mono & Mono SE use different mainboards.

jhbruhn, 2021-12-14:
Oops, you're right, but they seem to have similar headers on there. Maybe I will have to look at my machine and do some experiments.