luizribeiro/mariner #328: Saturn: Network Support

dirkessl, 2021-03-27:
It would be nice to have Network support for Chitu printers that have a network port. Chitu printers listen on port 3000/udp and you can send Gcode M commands through that.  e.g. _echo -n M27 | nc -w 1 -u 192.168.130.150 3000_

printers on the network can be found with _echo -n M99999 | socat - UDP4-DATAGRAM:192.168.130.255:3000,so-broadcast_

luizribeiro, 2021-03-27:
That's a good idea! Do you have access to one of those printers that has network?

I would need someone to take on this, since I don't have one myself.

It should be fairly easy to build a quick prototype of this. In order for us to ship it though, we would need some sort of configuration file for `mariner`, which we don't have yet.

dirkessl, 2021-03-28:
Yes, I have a Saturn. I tried to implement it myself, but didn't get it running on my computer (Ubuntu amd64) I guess a config file would be nice, but looking at the code, making an additional file (mars-network.py) should be enough for the prototype. Switching over to a network config can then be done by adding the network config to this file and renaming it.

Cheers,
Dirk

hansale, 2021-03-28:
Also of note is that the Chitusystems website has the module for adding wifi module support to their motherboards, that they sell (although I think it could be bought elsewhere
 https://www.chitusystems.com/product/esp-01-wifi-module/ 
Along with  tutorial on how to add the network printing functionality. 
https://www.chitusystems.com/2020/11/03/how-to-activate-the-wifi-module-function-on-your-3d-printer/

I assume this is the same functionality? The sonic mini 4k also has a slot where in theory you could solder an ethernet port.

dirkessl, 2021-03-29:
I found an unused raspberrypi and installed mariner3d on it and can therefore help with development/testing

luizribeiro, 2021-03-29:
I think we'll need the configuration file that it's being proposed on #329 here: so we can configure if the printer should be accessed through the serial port or through the network.

dirkessl, 2021-03-29:
Ok, with some socat magic, I have this running now on my Raspberry Pi 2 and Elegoo Saturn. There are some issues, though.

setup:

#create and connect together pseudo serial ports (on a fresh booted system this will be /dev/pts/2 and /dev/pts3
socat -d -d pty,raw,echo=0 pty,raw,echo=0

#forward the other end of the pseudo port to the network printer
socat -d -d -d -d /dev/pts/3 UDP4-DATAGRAM:192.168.130.58:3000

change the serial port in /opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/mars.py
self._serial_port.port = "/dev/pts/2"

restart mariner3d and you have mariner support without opening the printer.

The problem now is that mariner3d does not know the contents of the usb stick, so starting a print will fail if the contents of the usb stick does not match the contents of /mnt/usb_share

It would be possible to send the file over serial to the printer, but I am guessing this will be really slow. A solution could be to equip the raspberry zero-w with a network port via a ENC28J60 module. In this case it would not be needed to open the printer to get mariner3d working.

Also the Saturn has an internal 4GB disk which will be enabled with a future firmware update, so files could be put there.

[image]

[image]

figuerres, 2022-09-10:
is anyone working on this ?  i have a saturn and am thinking about how to get better remote access to manage printing.
is there a way to possibly use a regualr pi and a usb cable to make the pi hold the files and let the printer see the pi as a usb drive maby ?