luizribeiro/mariner #329: Make port number configurable

hansale, 2021-03-28:
I went into code and changed my port to 80, which of course then required root. I guess I could use something to forward the port on from port 80 that runs as apache, but having this port number be in a config file or allowing modification in some way might be ideal. I could probably make a PR for this if there's interest. I've got a list of about 20 minor enhancements that I've been writing down as I've been using this excellent program, after you said hi to me on HN after I gave you a shoutout on the Octoprint article. :+1:

hansale, 2021-03-28:
Feel free to assign quick hit issues to me @luizribeiro that you think are most pressing, I'd like to dive into the code base here, I've read and played about with the ctb file format thanks to your easy to read code, and have a good understanding of React and a decent understanding of Python although it's been a few years.

luizribeiro, 2021-03-29:
Sweet! I think this could be a good first option to have a configuration file.

I would love for us to use something like [`strictyaml`](https://pypi.org/project/strictyaml/) to define the config schema and allow configuring a few things through a file like `mariner.yaml` or something

I'm a bit swamped with work at the moment so don't have much time to dedicate to `mariner`, so if you're interested on taking on this this may be a good issue to work on

luizribeiro, 2021-05-09:
@hansale, I have added support for configuring mariner through a config file, `/etc/mariner/config.toml`. It's available from the `master` branch.

I imagine you could create a section for HTTP settings like this here:

```
[http]
port = 80
```

aef8b1ce0497ae0dc2e8c977ad5ebb365ed1ce3a is a good example of how to implement new configuration options.

luizribeiro, 2021-05-09:
Ended up building this :)

commit 9a6396cd (2021-05-10): Allow host/port to be configured

Fixes #329