luizribeiro/mariner #337: The printer returned an unexpected response: 'Error:File is not existed or empty, file "XYZ"

Zenroth, 2021-03-30:
So I'm not sure what causes this, but I've ran into it 3 times now, where I upload a file, I can open the file in preview panel it renders, etc., but then I select to print it and I get an exception.

Traceback (most recent call last):
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/server/api.py", line 208, in printer_command
    elegoo_mars.start_printing(filename)
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/mars.py", line 140, in start_printing
    raise UnexpectedPrinterResponse(response)
mariner.exceptions.UnexpectedPrinterResponse: Error:File is not existed or empty, file /crossbows2.ctb

It seems like if I reload the Mariner web interface it starts to work, but until than I start to continually get this error for any files uploaded and attempts to issue a print command.

luizribeiro, 2021-04-04:
It's probably not reloading that fixed the issue for you, but waiting until the printer "sees" the new file.

Which printer are you using?

Zenroth, 2021-04-05:
Sonic Mini 4k and Sonic Mighty 4k

luizribeiro, 2021-04-16:
The solution we're discussing on #311 will help with this, so going to close this.

closed 2021-04-16 by luizribeiro