luizribeiro/mariner #340: Elegoo Mars 2 Mono supported?

E-Olsen, 2021-03-31:
I was wondering if the green topped Mars 2 Mono was supported.  I have installed it, get 'printer ready' responses from it but am getting the following error when I try to print a file:

Traceback (most recent call last):
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/server/api.py", line 208, in printer_command
    elegoo_mars.start_printing(filename)
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/mars.py", line 132, in start_printing
    self.select_file(filename)
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/mars.py", line 111, in select_file
    raise UnexpectedPrinterResponse(response)
mariner.exceptions.UnexpectedPrinterResponse: //Disk init fail!

I have updated  and upgraded my Raspberian Buster and got the following error when it got to updating Mariner3d:

Processing triggers for mariner3d (0.1.1-1) ...
+ echo postinst called with unknown argument `triggered'
postinst called with unknown argument `triggered'
+ exit 1
dpkg: error processing package mariner3d (--configure):
 installed mariner3d package post-installation script subprocess returned error exit status 1
Processing triggers for install-info (6.5.0.dfsg.1-4+b1) ...
Processing triggers for initramfs-tools (0.133+deb10u1) ...
Processing triggers for ca-certificates (20200601~deb10u2) ...
Updating certificates in /etc/ssl/certs...
0 added, 0 removed; done.
Running hooks in /etc/ca-certificates/update.d...
done.
Processing triggers for libvlc-bin:armhf (3.0.12-0+deb10u1+rpt1) ...
Errors were encountered while processing:
 mariner3d
E: Sub-process /usr/bin/dpkg returned an error code (1)

So, I cannot print and the standard updating path is also not working.  I am using 5v @ 3.4a for my Pi Zero W so it should not be a power issue.  Any ideas?

E-Olsen, 2021-03-31:
I walked through the PI set up steps again, everything seems ok on the mariner install and I have v0.1.1-1 installed.

E-Olsen, 2021-04-01:
I tried a different file today and am getting the following:

[image]

E-Olsen, 2021-04-04:
Decided to start over with a larger SD card.  This is likely user error and should be resolved with my rebuild.

E-Olsen, 2021-04-04:
Rebuild complete, swapped the SD cards and tried again.  Got the following error:

Traceback (most recent call last):
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/server/api.py", line 208, in printer_command
    elegoo_mars.start_printing(filename)
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/mars.py", line 132, in start_printing
    self.select_file(filename)
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/mars.py", line 111, in select_file
    raise UnexpectedPrinterResponse(response)
mariner.exceptions.UnexpectedPrinterResponse: //Disk init fail!

E-Olsen, 2021-04-04:
dmesg gave me the following clues:

[   18.780058] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1   
[   18.780276] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available  
[   18.781502] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Oct 22 2019 01:59:28 version 7.45.98.94 (r723000 CY) FWID 01-3b33decd
[   24.524377] random: crng init done
[   24.524415] random: 7 urandom warning(s) missed due to ratelimiting
[   25.172779] 8021q: 802.1Q VLAN Support v1.8
[   27.396404] uart-pl011 20201000.serial: no DMA platform data
[   27.679040] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   28.010343] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
[   30.100007] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   30.348446] Mass Storage Function, version: 2009/09/11
[   30.348486] LUN: removable file: (no medium)
[   30.348708] lun0: unable to open backing file: /piusb.bim
[   30.348840] g_mass_storage 20980000.usb: failed to start g_mass_storage: -2   <-- error message

I went back through the configuration settings for the mass storage and copy/pasted them right from the instructions.  After a reboot, still have the same issue.  I am going to test this as a USB Mass Storage outside my Mars 2 and see if I can figure out what the issue is.

luizribeiro, 2021-04-04:
> [ 30.348708] lun0: unable to open backing file: /piusb.bim

You probably have a typo somewhere, the name of the image file should probably be `piusb.bin`, not `piusb.bim`

luizribeiro, 2021-04-11:
I'm going to close this issue since it doesn't seem like a bug, but rather misconfiguration on your end.

closed 2021-04-11 by luizribeiro

E-Olsen, 2021-04-11:
Can confirm, all is well now.  :)