luizribeiro/mariner #345: Samba upload

MSergW, 2021-04-05:
Hi!
how to allow uploading via smb?

from this manual - https://www.reddit.com/r/ElegooMars/comments/legz2i/saturn_rpi_mariner_samba/ - not upload - say that the file (empty.txt) is too large for the destination filesystem

hansale, 2021-04-07:
I’m running my setup far away from the WiFi and it hasn’t failed, for files 100+ MB. Maybe your microsd card is corrupted?

luizribeiro, 2021-04-11:
Yeah I generally upload 100+ MB files in a few seconds. I would check your wifi connection signal, try running with the Pi outside of the case, adding an external antenna to the Pi or other things like that.

I feel like it's unlikely that file upload would be much faster on Samba than over HTTP upload. I personally have had a lot of trouble in the past making Samba hit high transfer rates, depending on the device it's talking to.

You may want to upload the file over `scp` instead if you really want to try something other than the UI upload. You already have a SSH service running anyway :)

If you want to setup Samba I would recommend you to follow a tutorial like https://magpi.raspberrypi.org/articles/samba-file-server

It's a bit orthogonal to mariner though, so I'm going to close this, as it's neither an issue nor a feature request.

closed 2021-04-11 by luizribeiro