luizribeiro/mariner #346: Feature Request: Name Printer

Zenroth, 2021-04-06:
Simple one, probably would go with efforts to have config files. Being able to set a printer name would be quite helpful to us running multiple printers. (I've currently got 4 printers all running Mariner for instance)

PoetikDragon, 2021-04-06:
Seconded!

hansale, 2021-04-07:
The raspberry pi zero w uses mdns, for example I changed my host name for my printer to sonicmini4k1, that way I navigate to http://sonicminic4k1:5000 to see the interface. If people got in the habit of changing their host name, it’d be relatively easy to then put the host name in the title of Mariner

Zenroth, 2021-04-07:
> The raspberry pi zero w uses mdns, for example I changed my host name for my printer to sonicmini4k1, that way I navigate to http://sonicminic4k1:5000 to see the interface. If people got in the [...]

Interesting, I've never dug into it, but the host name stuff has never worked for me on windows with the raspberry pi zero w units over wifi, only when setup and connected via USB. (Not sure if this is a windows issue, and I need to install bonjour/etc, or if the multi-cast udp just isn't being forwarded across the wifi extender that the units are connected too.) 

I actually just modified my system hosts file the other day to work around this issue on a single system.

doucettom, 2021-04-16:
One can just change the title in index.html as well. I'm doing both hostname+title but I could probably pass the hostname as a variable for the title somehow.

luizribeiro, 2021-05-09:
I just implemented this. If you run mariner from the `master` branch (I haven't released a new version yet with this feature), you can configure the display name for the printer from `/etc/mariner/config.toml`.

closed 2021-05-09 by luizribeiro

doucettom, 2021-05-19:
How can I run master ?

luizribeiro, 2021-05-21:
You can download from the artifacts generated by continuous integration. For example, for the latest commit on master as of today: https://github.com/luizribeiro/mariner/actions/runs/845877861

commit 7fe50047 (2021-05-09): Allow for printer display name to be configured

Fixes #346