luizribeiro/mariner #360: File Upload Progress Bar

juicyroots, 2021-04-08:
Since the files we are uploading are large and the Pi Zero WiFi can be slow, it would be great to get a file upload progress bar. Yesterday, I was uploading a 150MB file, and after 5 minutes, I started to wonder if it was still in progress.

drashna, 2021-04-10:
Yes, please!

ixtenebrae, 2021-04-23:
What if you just changed CircularProgress here:
```
              isUploading ? <CircularProgress size={18} /> : <PublishIcon />
```
https://github.com/luizribeiro/mariner/blob/e4512bfb964b8ebf568078f19cb179699ab53da6/frontend/src/components/UploadButton.tsx#L61

to CircularProgressWithLabel or should it be a proper bar?

luizribeiro, 2021-07-09:
This has been addressed by #442

closed 2021-07-09 by luizribeiro