luizribeiro/mariner #365: '//Disk init fail!\r\n'

proffrink, 2021-04-09:
Going pretty mad with this one as I've tried everything. Three fresh installs and a scouring of the issue reports on here, but nothing. I've checked issues other issues:
#152 
#235 
#340 

I am essentially having the same issue as #152. When I run a print file, I get an error like this:
```
`Traceback (most recent call last):
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/server/api.py", line 208, in printer_command
    elegoo_mars.start_printing(filename)
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/mars.py", line 132, in start_printing
    self.select_file(filename)
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/mars.py", line 111, in select_file
    raise UnexpectedPrinterResponse(response)
mariner.exceptions.UnexpectedPrinterResponse: //Disk init fail!`
```

I believe this to be down to mounting the virtual USB but I'm not sure where I'm going wrong.

This is on a Phrozen 4K Mini (which I have on good authority works fine with this). My pinouts are good, hardware is fine (tried two boards, alternative power sources and get a different error when I remove the dupont jumpers, which indicates to me that this is an issue with how I've set up my Pis.

Here're my important files:
(Mostly checked against #152 so you might see a few small bits not in the official guide)

/boot/cmdline.txt
> `dwc_otg.lpm_enable=0 console=tty1 root=PARTUUID=b9509d4a-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait modules-load=dwc2 quiet splash`

/boot/config.txt
[...]

> `[all]
#dtoverlay=vc4-fkms-v3d

# Enable UART
dtoverlay=dwc2
start_x=1
enable_uart=1

gpu_mem=128`

/etc/fstab
> `proc            /proc           proc    defaults          0       0
PARTUUID=b9509d4a-01  /boot           vfat    defaults          0       2
PARTUUID=b9509d4a-02  /               ext4    defaults,noatime  0       1
# a swapfile is not a swap partition, no line here
#   use  dphys-swapfile swap[on|off]  for that
/piusb.bin /mnt/usb_share vfat users,gid=mariner,umask=000 0 2`

/etc/rc.local

> `#!/bin/sh -e
#
# rc.local
#
# This script is executed at the end of each multiuser runlevel.
# Make sure that the script will "" on success or any other
# value on error.
#
# In order to enable or disable this script just change the execution
# bits.
#
# By default this script does nothing.

# Print the IP address
_IP=$(hostname -I) || true
if [ "$_IP" ]; then
  printf "My IP address is %s\n" "$_IP"
fi

/bin/sleep 5

    modprobe g_mass_storage file=/piusb.bin removable=1 ro=1 stall=0 removable=1

    /sbin/iwconfig wlan0 power off

    exit 0`

proffrink, 2021-04-09:
df -h
(weird thing is the install script I used on this last-ditch effort - https://github.com/KTheMan/Mariner-Autoinstaller - should apparently have allocated 'all available space' to the virtual USB, but it looks like it's been allocated much less).
```
 Filesystem      Size  Used Avail Use% Mounted on
/dev/root        29G  5.5G   23G  20% /
devtmpfs        152M     0  152M   0% /dev
tmpfs           184M     0  184M   0% /dev/shm
tmpfs           184M  3.6M  180M   2% /run
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           184M     0  184M   0% /sys/fs/cgroup
/dev/loop0      4.0G  118M  3.9G   3% /mnt/usb_share
/dev/mmcblk0p1  253M   48M  205M  19% /boot
tmpfs            37M     0   37M   0% /run/user/1000
```

KTheMan, 2021-04-10:
> (weird thing is the install script I used on this last-ditch effort - https://github.com/KTheMan/Mariner-Autoinstaller - should apparently have allocated 'all available space' to the virtual USB, but it looks like it's been allocated much less).

Little bit of misinterpretation there. It's expanding the filesystem of Raspbian Lite to the whole microSD card, and not the virtual USB unfortunately.
If you're looking to add a larger amount of space to that virtual USB you can change the size line 116 (in the script) from its default of 4096 (which shows as 4.0G when you ran `df -h`)
`sudo dd bs=1M if=/dev/zero of=/piusb.bin count={Your_Virt_USB_Size_Here}`

proffrink, 2021-04-10:
> > (weird thing is the install script I used on this last-ditch effort - https://github.com/KTheMan/Mariner-Autoinstaller - should apparently have allocated 'all available space' to the virtual USB, but it looks like it's been allocated much less).
> 
> Little bit of misinterpretation there. It's expanding the filesystem of Raspbian Lite to the whole microSD card, and not the virtual USB unfortunately.
> If you're looking to add a larger amount of space to that virtual USB you can change the size line 116 (in the script) from its default of 4096 (which shows as 4.0G when you ran `df -h`)
> `sudo dd bs=1M if=/dev/zero of=/piusb.bin count={Your_Virt_USB_Size_Here}`
> 
> Have you tried plugging your micro USB end into the other port on your Pi Zero?

Good point, thank you. No, I've not tried that but I'll give it a go. Was just assuming headless mode so used power only on both boards that I tried. Will report back!

proffrink, 2021-04-10:
> Little bit of misinterpretation there. It's expanding the filesystem of Raspbian Lite to the whole microSD card, and not the virtual USB unfortunately.
> If you're looking to add a larger amount of [...]

Good point, thank you. No, I've not tried that but I'll give it a go. Was just assuming headless mode so used power only on both boards that I tried. Will report back!

proffrink, 2021-04-10:
No dice I'm afraid.

I'm actually going to try an entirely different machine today too just to try and rule that out.

proffrink, 2021-04-10:
Ok so it was a bad USB cable in the end (well, two of them). Much thanks to @KTheMan for his install script as I can see this making things much easier for people in the future.

closed 2021-04-10 by proffrink

drashna, 2021-04-10:
I'm running into the same issue.  Rebooting the pi fixed the issue for me, though.

503Dev, 2023-10-04:
To anyone who finds this in the future: it's probably your USB cable. I run a print farm. Have tons of these working. Just had one that gave me disk init failed. Used 4 different USB cables that I know work fine and have data lines. Same issue. 

Tried one other cable and it works fine instantly. Can easily repeat to confirm that was the fix. No idea why or what makes some cables special but all my cables have data and voltage lines and work fine in the farm.