luizribeiro/mariner #376: Peopoly Phenom L

MKronmiler, 2021-04-16:
I have an Elegoo Mars 2 Pro and an Peopoly Phenom L. I just finished successfully setting up the Mars 2 Pro with mariner and decided to take a peek at the Phenom L Control Board.

[image]

It looks like it's an unmodified [ChiTu L HDMI H1
](https://www.chitusystems.com/product/chitu-l-hdmi-h1/) with an exposed UART like the Mars 2 Pro. Is there anything stopping me from doing the same mariner install that I did on my Mars 2? My only concern is the Phenom L using a different board architecture that isn't supported.

luizribeiro, 2021-04-16:
I would imagine all these ChiTu boards are very similar to each other and it might work (do at your own risk though :)), but hard to know since I only own a Mars Pro.

Isn't that an ethernet port, btw? You might be able to get it to work with something similar to what is described on #328, without soldering anything

luizribeiro, 2021-04-16:
If you do end up trying this, let us know what happens (:

MKronmiler, 2021-04-18:
I found the open serial connection on the motherboard and followed the same process of installing and wiring up the Peopoly Phenom L. Everything appears to be working exactly as the Mars 2 Pro. I'll try to upload a guide for the Phenom L but the other thing that changes is the physical location of the open serial connection.

[image]

luizribeiro, 2021-04-18:
That's awesome! I'll update `README.md` with this as a supported printer.

commit e4512bfb (2021-04-18): Added Peopoly Phenom L to list of supported printers

Fixes #376
closed 2021-04-18 by luizribeiro