luizribeiro/mariner #388: Feature Request: job completion notification

rfield19, 2021-05-01:
Didn't see this as a suggestion - apologies if it has been already.

Suggestion for notification on completion of job - ie email / pushover/ etc.
Could have details pulled from object - thumbnail image, time taken to print, etc.

SebastianSchultz, 2021-05-06:
Yeah that would be great.
But I think that @luizribeiro see such a feature in an upcoming plugin system, right?!

luizribeiro, 2021-05-08:
If you use something like Home Assistant or Node-RED you should be able to set this up without any additional code on mariner. See my answer on #393

As to adding this functionality to mariner itself, I'm open to the idea. I think as long as we don't need to install additional hardware or software and it's a feature that most would want, it's okay to put it on the main distribution.

Modern browsers can use the [Notifications API](https://developer.mozilla.org/en-US/docs/Web/API/Notifications_API/Using_the_Notifications_API) to send notifications. This would be ideal. However, it's only available on sites that use HTTPS (which I imagine most folks who setup mariner don't have setup), so I don't think it's a good option.

For me personally email would be ideal, since I don't print often enough to want a push whenever it finishes. Maybe we could have a setting to use an SMTP server for these once we have a configuration file? I'm not sure if that would be helpful for people though.

Any other ideas?

rfield19, 2021-05-08:
Guessing having an smtp server setting would be straightforward enough - as the backend is already pretty much built into linux?

Maybe stick smtp functionality into the main build, and then leave more complex things like pushover, etc to the plugin system.

daxliniere, 2021-05-15:
All of this and more can be done with Node-RED which comes pre-installed on Raspberry Pi OS (including Light). I am actually in the process of building a platform to do all of this and more.
Definitely check out Node-RED, it's free, awesome and the community is very nice and helpful.

All the best!
Dax.
[image]

rfield19, 2021-05-15:
Thanks, yes I'd been trying out node-red this week - does seem powerful. 
I've installed it as a docker image on a NAS currently, which seems to work great.

At the moment I've got it sending out hourly job updates, and print completion via pushover api.  
I'm sure it can do a lot more though in the hands of someone more skilled!

daxliniere, 2021-05-15:
@rfield19, I didn't know anything about Node-RED until a month ago. :)
Though I am an engineer and can think logically, I'm definitely not a programmer, so I'm sure you can learn NR, too.

Once I've got my system all figured out, I'm going to post the entire flow to import directly into NR. Then you just need to select which sensors and output relays you have chosen to connect.