luizribeiro/mariner #400: Printer: EPAX E10/X10

rfield19, 2021-05-12:
I'm testing mariner on the EPAX E10 over the next few days.
Can't see at this stage why it wouldn't work well though, and be added to the supported printers list.

The board in the epax is an unmodified Chitu L M1 (32 bit tmc2209), and the serial pins are easily accessible.

The epax also has an active network port - which is fully supported by Chitubox, but unfortunately not by Lychee yet.  If network support for the Elegoo Saturn is added as per previous thread, highly likely it will also work for this.

[image]
[image]

rfield19, 2021-05-14:
OK, the EPAX E10 is now confirmed working - both over local usb to pi storage, and over network.
Another one for your compatibility list!

Very straightforward to mod as the epax motherboard has 2.54mm pins on the relevant serial connections already.
[image]
[image]

luizribeiro, 2021-05-15:
Awesome! Thanks for reporting :)

I'll leave this open until I update the `README.md`

commit cb583298 (2021-05-15): Add EPAX E10/X10 to list of supported printers

Fixes #400
closed 2021-05-15 by luizribeiro