luizribeiro/mariner #408: Mariner3D working on Raspberry Pi 3 A+

doucettom, 2021-05-19:
Just a note to mention that I was able to make mariner3D work on Raspberry Pi 3 A+. I realized that ustreamer was using 85@100% on the Pi Zero W so I gave the Pi 3 A+ a try. It's pretty much same process except for the config.txt entry:
dtoverlay=dwc2
becomes:
dtoverlay=dwc2,dr_mode=peripheral

Downside is it pulls off more current and heat more. Upside is 4 cores vs 1 :)

doucettom, 2021-05-21:
Oh and if you take the card from the pi zero, you'll have to re-add the wpa_supplicant.conf and modify /boot/config.txt as stated above :)

Video is a lot smoother and the web UI seems faster, it heats a tad more and I cannot power it from the motherboard's 5V pin like I did with the Pi Zero. I also taped the 5V pin of the USB connector as I wasn't sure it would fry something or not (I didn't taped on the zero)

luizribeiro, 2021-05-24:
Cool! I'm not sure what are all the places where we've documented the Pi Zero, but I'll try to find those and close this issue. Thanks for the report

rfield19, 2021-05-26:
Hi,
just a question / update for this before it's closed!  
I tried this out as the pi3 is quite a bit faster, but it disabled the ethernet port as soon as the 'peripheral' line is added to the config as above (which was a bit awkward as it was running headless!)

Despite trying multiple suggestions online, I've not been able to get both the ethernet port and the serial working at the same time.  Any suggestions on how to get it working?  I've tried the other two dwc2 modes but it doesn't help.

doucettom, 2021-05-26:
> Hi,
> just a question / update for this before it's closed!
> I tried this out as the pi3 is quite a bit faster, but it disabled the ethernet port as soon as the 'peripheral' line is added to the config [...]

You likely tried with a Pi 3B or a Pi 3B+. These won't work in usb gadget as they are integrating a USB hub to which the Ethernet port is linked. Only a 3A+ will work and the 3A+ doesn't have an Ethernet port (but has wifi)

rfield19, 2021-05-27:
Ah ok, thanks for the info - yes it was a 3b as I specifically wanted the ethernet.  
The pi zero(s) drop the wifi connection constantly so they're not really a solution for me.  Might try a 3A+ then and see if the wifi is more reliable.

doucettom, 2021-05-27:
> Ah ok, thanks for the info - yes it was a 3b as I specifically wanted the ethernet.
> The pi zero(s) drop the wifi connection constantly so they're not really a solution for me. Might try a 3A+ then and see if the wifi is more reliable.

Were you powering the pi from motherboard Pin ?

rfield19, 2021-05-27:
No, they're powered by a 2A mains adaptor.  No idea why it keeps dropping the connection on the pi zero - I've written cron scripts that ping an ip every minute to keep the connection open, disabled power management and so on - but it doesn't seem to make much difference.

doucettom, 2021-05-27:
> No, they're powered by a 2A mains adaptor. No idea why it keeps dropping the connection on the pi zero - I've written cron scripts that ping an ip every minute to keep the connection open, disabled power management and so on - but it doesn't seem to make much difference.
Mines were fine from the printer's board but it looks like yours might have power problems. You might want to give them a try with a 2.1@2.5A supply just in case. I used a 5.2v 2.1A charger last time on a pi 1b+ on another project and so far so good... My 3A+ wasn't happy with a 2.4A phone charger or the printers 5v pin though and it exhibited the same behavior your zero was doing (wifi drops and serial not working)

swapping the PSU for an official 2.5a pi supply fixed all of this !

rfield19, 2021-05-27:
I have an official pi psu I use for a pi3, so will give that a go.  
Buying a new psu is cheaper than buying a pi3a+ so fingers crossed.  

If it does work, at least the psu info might be useful for other mariner users!

luizribeiro, 2021-05-28:
@doucettom, I'm actually surprised that you have this running on the Pi R3A+. Are you sure that's the model you have?

The RPi 3A+ does **not** have USB OTG: https://en.wikipedia.org/wiki/Raspberry_Pi#Specifications

This means that it shouldn't be able to act as a USB Gadget and share files with the printer with `g_mass_storage`. Either the Wikipedia article is wrong or you have a magic RPi 🙃

I have updated the documentation to include the RPi 4B, since that model has USB OTG and we've had reports of it being used successfully. See https://mariner.readthedocs.io/en/latest/getting-started.html

Let me know and feel free to put up a pull request to edit the documentation once you have confirmed this.

doucettom, 2021-05-29:
> @doucettom, I'm actually surprised that you have this running on the Pi R3A+. Are you sure that's the model you have?
>
> The RPi 3A+ does **not** have USB OTG: [...]

[image]

You are right with the Pi 3A+ not supporting the OTG feature. What OTG does is switching between host and gadget automatically. That being said the Pi 3 A+ can be set to be either host or gadget thus the:
dtoverlay=dwc2,**dr_mode=peripheral** to force it to be a peripheral instead of OTG
I think taping the 5V pin is also mandatory as a true OTG like the Pi Zero will also take care of this I think but it might have some consequences if both VCC pin is sending current. Some information I used:

https://www.raspberrypi.org/forums/viewtopic.php?t=228267

luizribeiro, 2021-05-29:
> I'd do a pull request if I knew how ;)

It's your lucky day! I wrote some documentation on how to make contributions here: https://mariner.readthedocs.io/en/latest/contributing.html

doucettom, 2021-06-04:
> > I'd do a pull request if I knew how ;)
> 
> It's your lucky day! I wrote some documentation on how to make contributions here: https://mariner.readthedocs.io/en/latest/contributing.html

I'll try to get this done this weekend

doucettom, 2021-06-10:
Solved and completed with:
https://github.com/luizribeiro/mariner/pull/421/
https://github.com/luizribeiro/mariner/pull/422

closed 2021-06-10 by doucettom