luizribeiro/mariner #412: Error: Disk init fail!

bludin, 2021-05-28:
Thanks for mariner3d! What a great project.
I've just installed it on a RPi0W and a Phrozen Sonic Mini 4K following the instructions in the blog.  The web interface responds, uploading files work, but when trying to print, I get:

```
Unexpected Printer Response
The printer returned an unexpected response: '//Disk init fail!\r\n'

Traceback (most recent call last):
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/server/api.py", line 220, in printer_command
    elegoo_mars.start_printing(filename)
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/mars.py", line 132, in start_printing
    self.select_file(filename)
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/mars.py", line 111, in select_file
    raise UnexpectedPrinterResponse(response)
mariner.exceptions.UnexpectedPrinterResponse: //Disk init fail!
```

the log doesn't show anything peculiar

```
pi@sm4k:~ $ sudo journalctl -u mariner3d
-- Logs begin at Fri 2021-05-28 13:32:27 CEST, end at Fri 2021-05-28 13:44:42 CEST. --
May 28 13:32:42 sm4k systemd[1]: Started mariner3d Printer Controller.
May 28 13:32:56 sm4k mariner[273]: INFO:waitress:Serving on http://0.0.0.0:5000
```

Any idea what's wrong?
Thanks for your help,  beat

bludin, 2021-05-28:
ok, I found the solution myself:

the instructions say:
```
Finally, make /etc/rc.local load the g_mass_storage module by adding this to it:

#!/bin/sh -e
modprobe g_mass_storage file=/piusb.bin stall=0 ro=1
exit 0
```

However, the existing dummy script exits before executing the new lines, so just appending them won't work. You either have to comment out our delete the dummy script (probably just commenting out the previous 'exit 0' will do, too, but I haven't tested it)

luizribeiro, 2021-05-28:
You shouldn't have to comment out or delete any existing scripts. Just adding `modprobe g_mass_storage file=/piusb.bin stall=0 ro=1` to it is sufficient.

None of the packages on Raspberry Pi OS Lite provide a `/etc/rc.local`, so in most cases you can just add the file with the contents from the instructions.

Anyway, I probably assumed some knowledge about shell scripting there when writing those instructions. What do you think would make this clearer?

I'll probably add a troubleshooting section too to our docs, so I'm going to leave this open. Would love some help to fill the troubleshooting section once I have that setup.

luizribeiro, 2021-05-29:
FYI I added a Troubleshooting section to the documentation, including instructions on how to troubleshoot the "Disk init fail" error: https://mariner.readthedocs.io/en/latest/troubleshooting.html

I also updated the software installation instructions so it's clearer that the contents mentioned are only applicable if the file does not exist: https://mariner.readthedocs.io/en/latest/software-setup.html

If the after the updates the steps aren't clear enough yet let me know!

commit a422b0df (2021-05-29): [docs] Make steps to update rc.local clearer

Fixes #412
closed 2021-05-29 by luizribeiro