luizribeiro/mariner #414: run .gcode files from web interface

bludin, 2021-05-30:
I can run ("print")  .gcode files from the UI of my Sonic Mini 4K, but when I try the same from the mariner3d web interface, I get a _'this file cannot be printed'_ error.

luizribeiro, 2021-05-30:
Yeah we could allow for `.gcode` files to be executed from the mariner UI. Out of curiosity, do you need to do this often enough that it's worth doing it remotely?

bludin, 2021-05-30:
That's hard to say. I'm just starting out. I guess I could just use the console for that purpose, but so far, I haven't had any success issuing gcodes/mcodes via minicom.

luizribeiro, 2021-05-30:
Why not just copy the `gcode` file to `/mnt/usb_share` and hit print on the printer itself? From my experience it's pretty rare for you to want to run `gcode` files on a MSLA printer. I only did it once when I initially setup my printer settings (disabling FAN when printer isn't running, etc). So I'm not sure if this is very high pri unless there's a use case I'm missing.

bludin, 2021-05-30:
You've certainly got a point there. It's just that I'm often 40km away from the printer and currently in the phase of trying things out while it's not printing. I guess it's a question of effort vs. usefulness. I assumed you might just have to change a line or two of code.

drashna, 2021-06-05:
I've seen "mix resin in vat" gcodes.  So ... it could very well be useful to have.

dprossner, 2022-11-08:
@luizribeiro 
Apologies for trying to resurrect this after almost a year, but adding an exception for .gcode files would be incredibly helpful. Ideally, an "at your own risk" mode where Mariner will pass through any file without trying to analyze them and keep layer progress would also be amazing as it could help facilitate firmware updates remotely.

In my current situation, I have my printers in a garden shed and they often sit for a while in between prints. right now I still have to physically walk into the shed before starting my prints to press the button on the printer to run my vat mixing gcode to make sure pigments that have fallen out of suspension get reincorporated.

My situation is likely an edge case but I can't imagine that having resin printers physically isolated from living spaces is all that unique.

I would love to collaborate on this and while I'm not particularly code-savvy, I would be more than happy to help with testing.