luizribeiro/mariner #420: Concise description here

tarantulus, 2021-06-07:
**Description**
500 Error on listing files

**Reproduction Steps**
Install following instructions
Attempt to view file list

**Expected Behavior**
see a list of files

**Screenshots / Logs**
[image]
[image]

**Details:**
 - Mariner Version: 0.2.0-1
 - Printer Model: Elegoo Mars
 - Raspberry Pi Model: RPi Zero W

**Additional context**
Uploads seem to work, but the printer is unable to see files either

luizribeiro, 2021-06-12:
@tarantulus, do you have a bit more of information about what is on your `/mnt/usb_share`?

Some thoughts here:

1. A `MemoryError` means that the program ran out of memory
2. It's possible your Pi doesn't have enough free memory due to another process that is running
3. It's possible that you may have a lot of files under `/mnt/usb_share` (or some files that are causing mariner to consume an excessive amount of memory due to a bug)

Can you investigate those things and report back your findings? You can check which process is using most memory with `htop` for example. Getting a list of files (including sizes?) that you have under `/mnt/usb_share` could help too.

tarantulus, 2021-06-22:
htop shows <1% memory usage
/mnt/usb_share has one 54mb file

possibly related to https://github.com/luizribeiro/mariner/issues/433

tarantulus, 2021-06-22:
After further testing, this is definitely related to #433. Files from chitubox do not cause this issue.

luizribeiro, 2021-06-22:
Sounds good, I'll close this in favor of #433 then. Thanks @tarantulus!

closed 2021-06-22 by luizribeiro