luizribeiro/mariner #424: Can't upload same file twice

daxliniere, 2021-06-11:
Upload a file called "xyz.ctb"
Delete that file.
Upload another file called "xyz.ctb"
RESULT: File will not upload until you refresh the page.

CI build of Mariner3D from about 5 days ago(?) on RPi Zero W

luizribeiro, 2021-06-12:
Thanks! I can repro this sometimes. Since there's no network request being sent when that happens it seems like the issue is on the frontend somewhere.

morairtym, 2021-06-29:
Similar thing. Not the first time I have had this happen.  Should be on latest release as I set it up in June.
but mine seems to lock up after trying to reupload the file
It could be a chrome extension, but I'll test in incognito mode after this print finishes.

But steps are like this
Upload file
Delete file
File still in UI
Refresh page
File gone
Upload
File could not be uploaded error. Cant remember the exact error but if it happens on a retest in incognito I'll let you know.
Then web interface crashes and I need to reboot. 
After that I can upload just fine.

amd989, 2026-11-11:
[Mariner 1.0.0](https://github.com/mariner3d/mariner/releases/tag/v1.0.0) replaced the web interface, including the upload flow this was about. Could you try 1.0.0 and say whether this still happens? I'll close this in 30 days if there's no answer.

daxliniere, 2026-11-11:
Hey Alejandro.
Thank you, but I haven't even switched on my 3D printer for about 3 years.
Best of luck, though, Mariner is an awesome bit of software.