luizribeiro/mariner #425: Improve Hardware setup section of documentation

luizribeiro, 2021-06-12:
The [Hardware Setup](https://mariner.readthedocs.io/en/latest/hardware-setup.html) section of the documentation is pretty basic.

Ideally we would have better instructions on how to do the hardware setup, so it's more friendly to folks who don't understand much about electronics.

This could include pictures and, potentially, separate instructions for each printer on separate documentation pages. Though since the instructions for most printers is largely the same, I'm not sure it's worth it. We could also keep printer-specific documentation under the wiki on GitHub.

If you'd like to take on this issue, I recommend reading the Contributing section of our docs that specific talks about how to contribute to Documentation: https://mariner.readthedocs.io/en/latest/contributing.html#documentation

rudetrooper, 2021-07-15:
[image]

luizribeiro, 2021-07-15:
@rudetrooper, would you like to take on this issue?