luizribeiro/mariner #432: Add option to add Authentication

m10653, 2021-06-17:
**Add option to add Authentication**
There should be an option to add at least basic auth so not anyone on the local network is able to control prints and potently cause damage.

We may want to only limit this to actions like uploading and start and stop. Allowing anyone to view the printer status? Or add that as an option to allow.

luizribeiro, 2021-06-18:
I'm not sure I would like us to support this.

I personally my mariner instance running behind a reverse proxy (nginx) with [oauth2-proxy](https://github.com/oauth2-proxy/oauth2-proxy) for handling authentication, as I didn't want to have separate credentials for this and just wanted to use OAuth instead. But you could even use nginx's own [basic HTTP authentication](https://docs.nginx.com/nginx/admin-guide/security-controls/configuring-http-basic-authentication/).

mariner is just a web application for controlling the 3D printer... If you need authentication I'd recommend you to use software that actually does the job well instead of having something that is half-baked on mariner.

DBa2016, 2021-07-19:
I would recommend using a reverse proxy (nginx, haproxy, apache, pick one) and implementing authentication there. Though it does not allow real access control (like readonly for everyone, read/write for authenticated or similar), I'd consider it an overkill for now, to be honest.
Mariner does not support HTTPS anyway, so every authentication method in it itself would be no more than fiction. If you want to secure it, limit listening to 127.0.0.1 and use a reverse proxy with whatever level of hard authentication you want.

luizribeiro, 2021-07-22:
Yeah, I'm going to close this. If anyone wants to add a section to the documentation explaining this at a high-level I'd love to review a pull request :)

closed 2021-07-22 by luizribeiro