luizribeiro/mariner #433: .cbddlp files from Lychee slicer cause 500 errors

dprossner, 2021-06-20:
**Description**
whenever I upload a .cbddlp file sliced in lychee slicer, the server throws a HTTP 500 error when I go to the files tab. the files print fine off of a usb drive but seem to crash mariner. .cbddlp files from chitubox and ctb files from both lychee and chitubox still work.

**Reproduction Steps**
Steps to reproduce the behavior:
1. slice a model in lychee slicer
2. upload it to mariner via the webui or SMB share
3. See error

**Details:**
 - Mariner Version: 0.2.0-1
 - Printer Model: Elegoo Mars
 - Raspberry Pi Model: Raspberry Pi Zero W

luizribeiro, 2021-06-21:
Do you have an example of one of those files? mariner is probably failing to parse it for some reason

dprossner, 2021-06-21:
> Do you have an example of one of those files? mariner is probably failing to parse it for some reason

GitHub doesn't seem to support the cbddlp filetype but [here's ](https://mega.nz/folder/dCI20LTK#uNfhZGJvWYDNSOZEbkRG-g)a link to files of the same model from both slicers.

wizard311, 2022-05-07:
I can verify  .cbt file created by lychee slicer cause the 500 Error.  My cbt. files created by lychee slicer are .cbt (v4) for use on Elegoo Saturn S.

amd989, 2026-11-11:
Fixed in [Mariner 1.0.0](https://github.com/mariner3d/mariner/releases/tag/v1.0.0): the file list no longer fails when one file can't be read. A file Mariner can't parse is listed as not printable instead of the whole page returning a 500, and encrypted `.ctb` files from newer ChiTuBox and Lychee versions are now supported.

If this still happens on 1.0.0, comment here with the log and I'll reopen.

closed 2026-11-11 by amd989