luizribeiro/mariner #444: The printer returned an unexpected response: ''

DBa2016, 2021-07-11:
**Description**
Mid-print I am seeing `The printer returned an unexpected response: ''`

**Reproduction Steps**
Connected the Saturn to a Raspberry PI, started the print.
Every now and then, the error message appears in the web interface. The print does not stop, and if I click on "OK", then at some point in time, mariner resumes status updates.
**Expected Behavior**
Normal print process.
**Screenshots / Logs**
Not sure where to find the logs.
**Details:**
 - Mariner Version: 0.2.0-1
 - Printer Model: Elegoo Saturn
 - Raspberry Pi Model: RPi Zero W

OSPFNeighbour, 2021-07-22:
Yes I believe I am getting the same issue, idle tabs popup am error but a page refresh brings it all back no problems.

Happy to assist in what ever debug is needed

luizribeiro, 2021-07-22:
This is the same as #180, it's been mostly fixed on master but we haven't released a new version yet (see #180 for details on how to get a `.deb` package from CI).

Closing the issue since it's a duplicate

closed 2021-07-22 by luizribeiro

nini57, 2021-07-27:
Hello,
After the last update with the last .deb I've again the same error

The printer returned an unexpected response: 'ok N:80\r\n'

How I can solve this ?
Thank's in advance