luizribeiro/mariner #452: The printer returned an unexpected response: 'ok N:160\r\n'  Details

nini57, 2021-08-12:
Hello,
i updated marinate to the latest version but i still have the UnexpectedPrinterResponse error: ok

Traceback (most recent call last):
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request ()
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions [rule.endpoint] (** req.view_args)
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/server/api.py", line 46, in print_status
    selected_file = elegoo_mars.get_selected_file ()
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/mars.py", line 102, in get_selected_file
    self._extract_response_with_regex ("ok '([^'] +) '\ r \ n", data) .group (1)
  File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/mars.py", line 39, in _extract_response_with_regex
    raise UnexpectedPrinterResponse (data)
mariner.exceptions.UnexpectedPrinterResponse: ok N: 160

do you have an idea
have a good day

nini57, 2021-08-30:
Hello,
Does somebody have an idea please.

Thanks in advance

tarantulus, 2021-08-31:
Are you sure your wiring is correct?

luizribeiro, 2021-09-21:
Is this preventing you from printing or is it just showing this message in the UI?

See "Unexpected Printer Response" under https://mariner.readthedocs.io/en/latest/troubleshooting.html#common-issues to check if it's the same issue mentioned there

nini57, 2021-09-24:
> Is this preventing you from printing or is it just showing this message in the UI?
> 
> See "Unexpected Printer Response" under https://mariner.readthedocs.io/en/latest/troubleshooting.html#common-issues to check if it's the same issue mentioned there

it just displays the error message. I manage to print

nini57, 2021-12-01:
> Hello all, I have exactly the same error message. I have checked the wiring un the Firmware is V4.4.3-c2 F9.23 . If you need more information contact me please ... :)

I still have the problem, you too?

nini57, 2022-01-03:
Hello,
I still have the same problem do you have a solution ?
best regards

amd989, 2026-11-11:
[Mariner 1.0.0](https://github.com/mariner3d/mariner/releases/tag/v1.0.0) reworked how Mariner reads the printer's status replies: commands are retried, buffers flushed, and M4000 replies parsed more tolerantly. Could you try 1.0.0 and say whether this still happens? I'll close this in 30 days if there's no answer.