luizribeiro/mariner #454: Something went wrong The server replied with a 500 HTTP status code.

chris1601, 2021-08-16:
Hi there,

I'm quite new to mariner so pleas be gentle to me 😊
I'm using the software on a Pi Zero W and a Mars Pro.

Mariner worked some prints (may 5 to 6) but the I got prints which seemed to be aborted at about half of the print time. However Mariner and the Mars seemed to be done with the print in the morning.
I tried the exact Chitufile directly from an USB-Stick and the print worked.

So I left Mariner alone for some time and now every time I start a print I get the message mentioned in the title. When I'm not printing I can access mariner just fine and can upload files.

Is there anything I can try or provide to help someone help me trying to figure this one out? 

And help or a pointer in the right direction would be much appreciated. 

Thanks guys and keep up the great work. 

sudo journalctl -ub mariner3d.service gives only:
Failed to add match 'mariner3d.service': Invalid argument 

sudo dmesg:
`Linux raspberrypi 5.10.52+ #1440 Tue Jul 27 09:53:08 BST 2021 armv6l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Mon Aug 16 17:03:53 2021 from 192.168.1.26
pi@raspberrypi:~ $ sudo dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.10.52+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1440 Tue Jul 27 09:53:08 BST 2021
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache[    0.000000] OF: fdt: Machine model: Raspberry Pi Zero W Rev 1.1
[    0.000000] Memory policy: Data cache writeback
[    0.000000] Reserved memory: created CMA memory pool at 0x17c00000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000] On node 0 totalpages: 114688
[    0.000000]   Normal zone: 1008 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 114688 pages, LIFO batch:31
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 113680
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:21:86:21 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  console=tty1 root=PARTUUID=2fd3879f-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait modules-load=dwc2
[    0.000000] Kernel parameter elevator= does not have any effect anymore.
               Please use sysfs to set IO scheduler for individual devices.
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 374400K/458752K available (8680K kernel code, 1324K rwdata, 2824K rodata, 424K init, 837K bss, 18816K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] ftrace: allocating 30707 entries in 60 pages
[    0.000000] ftrace: allocated 60 pages with 4 groups
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] random: get_random_bytes called from start_kernel+0x324/0x560 with crng_init=0
[    0.000024] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000095] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000181] bcm2835: system timer (irq = 27)
[    0.000853] Console: colour dummy device 80x30
[    0.001518] printk: console [tty1] enabled
[    0.001614] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
[    0.060347] pid_max: default: 32768 minimum: 301
[    0.060648] LSM: Security Framework initializing
[    0.060974] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[... 203 lines cut ...]
[   26.419370] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
[   26.663269] uart-pl011 20201000.serial: no DMA platform data
[   27.987243] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   31.182262] Bluetooth: Core ver 2.22
[   31.182449] NET: Registered protocol family 31
[   31.182466] Bluetooth: HCI device and connection manager initialized
[   31.192002] Bluetooth: HCI socket layer initialized
[   31.192040] Bluetooth: L2CAP socket layer initialized
[   31.192106] Bluetooth: SCO socket layer initialized
[   31.219514] Bluetooth: HCI UART driver ver 2.3
[   31.219544] Bluetooth: HCI UART protocol H4 registered
[   31.219690] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   31.219980] Bluetooth: HCI UART protocol Broadcom registered
[   32.572156] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   32.572181] Bluetooth: BNEP filters: protocol multicast
[   32.572214] Bluetooth: BNEP socket layer initialized
[   34.988967] Mass Storage Function, version: 2009/09/11
[   34.989003] LUN: removable file: (no medium)
[   34.989195] LUN: read only file: /piusb.bin
[   34.989213] Number of LUNs=1
[   34.991844] g_mass_storage gadget: Mass Storage Gadget, version: 2009/09/11
[   34.991876] g_mass_storage gadget: userspace failed to provide iSerialNumber
[   34.991890] g_mass_storage gadget: g_mass_storage ready
[   34.991911] dwc2 20980000.usb: bound driver g_mass_storage
pi@raspberrypi:~ $`

tarantulus, 2021-08-31:
Which slicer are you using?

TMillesich, 2021-09-18:
I can reproduce that HTTP 500 error.

Looks like mariner does not like it when you use an USB-stick instead of the internal storage area. 
I wanted to use mariner just to check the actual print status (actual layer/time left), this might not be possible.

sudo journalctl -n 200 -f
`
Sep 19 00:09:00 3dpi mariner[316]: FileNotFoundError: [Errno 2] No such file or directory: '/mnt/usb_share/_elderreddragon-body-head-tail_1-tail_2.ctb'
Sep 19 00:10:00 3dpi mariner[316]: ERROR:waitress:Exception while serving /api/print_status
Sep 19 00:10:00 3dpi mariner[316]: Traceback (most recent call last):
Sep 19 00:10:00 3dpi mariner[316]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/waitress/channel.py", line 397, in service
Sep 19 00:10:00 3dpi mariner[316]:     task.service()
Sep 19 00:10:00 3dpi mariner[316]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/waitress/task.py", line 168, in service
Sep 19 00:10:00 3dpi mariner[316]:     self.execute()
Sep 19 00:10:00 3dpi mariner[316]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/waitress/task.py", line 434, in execute
Sep 19 00:10:00 3dpi mariner[316]:     app_iter = self.channel.server.application(environ, start_response)
Sep 19 00:10:00 3dpi mariner[316]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/app.py", line 2464, in __call__
Sep 19 00:10:00 3dpi mariner[316]:     return self.wsgi_app(environ, start_response)
Sep 19 00:10:00 3dpi mariner[316]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/whitenoise/base.py", line 85, in __call__
Sep 19 00:10:00 3dpi mariner[316]:     return self.application(environ, start_response)
Sep 19 00:10:00 3dpi mariner[316]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/app.py", line 2450, in wsgi_app
Sep 19 00:10:00 3dpi mariner[316]:     response = self.handle_exception(e)
Sep 19 00:10:00 3dpi mariner[316]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/app.py", line 1867, in handle_exception
Sep 19 00:10:00 3dpi mariner[316]:     reraise(exc_type, exc_value, tb)
Sep 19 00:10:00 3dpi mariner[316]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
Sep 19 00:10:00 3dpi mariner[316]:     raise value
Sep 19 00:10:00 3dpi mariner[316]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
Sep 19 00:10:00 3dpi mariner[316]:     response = self.full_dispatch_request()
Sep 19 00:10:00 3dpi mariner[316]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
Sep 19 00:10:00 3dpi mariner[316]:     rv = self.handle_user_exception(e)
Sep 19 00:10:00 3dpi mariner[316]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception
Sep 19 00:10:00 3dpi mariner[316]:     reraise(exc_type, exc_value, tb)
Sep 19 00:10:00 3dpi mariner[316]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
Sep 19 00:10:00 3dpi mariner[316]:     raise value
Sep 19 00:10:00 3dpi mariner[316]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
Sep 19 00:10:00 3dpi mariner[316]:     rv = self.dispatch_request()
Sep 19 00:10:00 3dpi mariner[316]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
Sep 19 00:10:00 3dpi mariner[316]:     return self.view_functions[rule.endpoint](**req.view_args)
Sep 19 00:10:00 3dpi mariner[316]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/server/api.py", line 54, in print_status
Sep 19 00:10:00 3dpi mariner[316]:     FILES_DIRECTORY / selected_file
Sep 19 00:10:00 3dpi mariner[316]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask_caching/__init__.py", line 952, in decorated_function
Sep 19 00:10:00 3dpi mariner[316]:     rv = f(*args, **kwargs)
Sep 19 00:10:00 3dpi mariner[316]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/server/utils.py", line 20, in read_cached_sliced_model_file
Sep 19 00:10:00 3dpi mariner[316]:     return file_format.read(FILES_DIRECTORY / filename)
Sep 19 00:10:00 3dpi mariner[316]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/file_formats/ctb.py", line 131, in read
Sep 19 00:10:00 3dpi mariner[316]:     with open(str(path), "rb") as file:
Sep 19 00:10:00 3dpi mariner[316]: FileNotFoundError: [Errno 2] No such file or directory: '/mnt/usb_share/_elderreddragon-body-head-tail_1-tail_2.ctb'
`

luizribeiro, 2021-09-21:
Yup if you want to use an external USB drive you need to make tweaks. I haven't done it myself so I don't know the exact steps you would have to make though.

If there's enough interest for this we can probably add an option to the config. Feel free to create a feature request if you want to see this happen and maybe someone will pick it up

closed 2021-09-21 by luizribeiro