luizribeiro/mariner #478: files from ChiTuBox slicer (1.9 basic) cause 500 errors

JoshuaSKane, 2021-11-02:
I've set up mariner on a Raspberry Pi W 2, and a Phrozen Mega8K.

The setup went without error, and I was able to both connect to the printer via the Rx and Tx connections as well as upload files via the network share.   

Once I place one of the .CTB Files on the network share it errors out (Error 500 HTTP).  I know they added encryption to the latest version of ChiTuBox and I wonder if this is the root of the problem

luizribeiro, 2021-11-02:
I haven't upgraded my chitubox in a while and I don't really plan on doing that right now... Can you share an example file?

JoshuaSKane, 2021-11-03:
Here is a link to the file.  Sorry for the large size.  Let me know if you need any logs as well. 

https://drive.google.com/file/d/1XlS-GAviC8qAVWSOTrXD6onCuDfYE_Vm/view?usp=sharing

JoshuaSKane, 2021-11-03:
I did a but more looking into this. It looks like the file can be transferred without issue to the USB Share.   I then took the Pi Zero w 2 and hooked it up to the printer, but powered the pi externally so only the Tx and Rx were connected.  This is when I again ran into the 500 HTTP error.

JoshuaSKane, 2021-11-04:
I was also able to reproduce this on a Raspberry Pi 4.  It's the same HTTP 500 error.   Oddly enough when I was done with that I loaded the Pi Zero back up, connecting power, ground Rx and Tx and plugged in the USB.   It could read the file that was previously saved to the Zero, and I could pull up it's properties.  I went and remotely rebooted the Zero and I was back to the HTTP 500 errors.

luizribeiro, 2021-11-06:
Can you share mariner's logs? Are you sure this isn't the same as #453 ?

JoshuaSKane, 2021-11-07:
I’ve tried pulling the logs but I can’t find them. I know you call out the log file in your troubleshooting guide, but it tells me the file does not exist. Is there a location o can look, or a way to force the log to generate?

luizribeiro, 2021-11-07:
Did you try running `sudo journalctl -ub mariner3d.service` or `sudo systemctl status mariner3d.service`?

JoshuaSKane, 2021-11-07:
I can use sudo systemctl status mariner3d.service

This is what I get...

● mariner3d.service - mariner3d Printer Controller
   Loaded: loaded (/lib/systemd/system/mariner3d.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2021-11-05 13:17:06 PDT; 1 day 19h ago
 Main PID: 303 (mariner)
    Tasks: 6 (limit: 871)
   CGroup: /system.slice/mariner3d.service
           └─303 /opt/venvs/mariner3d/bin/python /usr/bin/mariner

Nov 07 07:17:39 Mega8K mariner[303]:     path / dir_entry.name
Nov 07 07:17:39 Mega8K mariner[303]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/flask_caching/__init__.py", line 952, in decorated_function
Nov 07 07:17:39 Mega8K mariner[303]:     rv = f(*args, **kwargs)
Nov 07 07:17:39 Mega8K mariner[303]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/server/utils.py", line 20, in read_cached_sliced_model_file
Nov 07 07:17:39 Mega8K mariner[303]:     return file_format.read(FILES_DIRECTORY / filename)
Nov 07 07:17:39 Mega8K mariner[303]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/file_formats/ctb.py", line 135, in read
Nov 07 07:17:39 Mega8K mariner[303]:     ctb_slicer = CTBSlicer.unpack(file.read(CTBSlicer.get_size()))
Nov 07 07:17:39 Mega8K mariner[303]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/typedstruct.py", line 89, in unpack
Nov 07 07:17:39 Mega8K mariner[303]:     return cls(*unpack(cls.get_format(), buffer))
Nov 07 07:17:39 Mega8K mariner[303]: struct.error: unpack requires a buffer of 76 bytes

JoshuaSKane, 2021-11-09:
I am beginning to think that this may be related to the ChituBox encryption they implemented on their files and boards. If I load a .CTB file that I sliced with 1.8 prior to the encryption it shows up, and lets me see the file, the thumbnails, layers and other information.  If I use a file sliced with 1.9 or pro it throws the 500 error.  There is a Chitu SDK (https://www.chitubox.com/en/page/sdk) That you can apply for to work with their new encryption.  This may also be similar to what was seen in issue [#453](https://github.com/luizribeiro/mariner/issues/453) as that firmware calls out (Mars C: Latest Firmware V4.4.1-F2.14 Update log: Compatible with Chitubox Pro and Chitubox 1.9.0)

Oddly I have an Elegoo Saturn with firmware  4.4.3 that adds the same support for Chitubox Pro and Chitubox 1.9 and Mariner is working well with that machine.

fcollingwood, 2021-11-12:
Liberating Mars can convert v3 (unencrypted) .ctb file to v4 (Encrypted) .ctb files

I've had a brief look at the Liberating Mars source code and modified it to add two stage lift. Possibly decryption can be added to Mariner based on the Liberating Mars encryption.

JoshuaSKane, 2021-11-12:
How can I help you with these efforts?

fcollingwood, 2021-11-12:
> How can I help you with these efforts?

I'm hoping Luiz can do it - I'm not to clued on Python. I'll dive into it if I have to, but it could take a while

fcollingwood, 2021-12-12:
@JoshuaSKane try this fork: https://github.com/BlueFinBima/mariner

Typhoan, 2022-01-10:
@fcollingwood Im not super familiar with linux. Do you have a link or some documentation i can use to compile/install of mariner?

fcollingwood, 2022-01-16:
> @fcollingwood Im not super familiar with linux. Do you have a link or some documentation i can use to compile/install of mariner?

Sorry, no clue.

koshisan, 2022-02-12:
@fcollingwood Does the mentioned fork contain changes to the file reading at all? I skipped over the changes and couldn't find anyhting that looks like it would...
(I have the fork running on my box here... - although it throws the same errors with my Mars 3 as the original version)

fcollingwood, 2022-02-13:
> @fcollingwood Does the mentioned fork contain changes to the file reading at all? I skipped over the changes and couldn't find anyhting that looks like it would... (I have the fork running on my box here... - although it throws the same errors with my Mars 3 as the original version)

No it doesn't.

I've hacked ctb.py to handle both encrypted and unencrypted, but likely won't be doing a pull request for it

koshisan, 2022-03-01:
@fcollingwood Any chance to change your mind about this? I must confess I am a little desperate... I even tried contracting a freelance developer to get it implemented and submit an official pull request, but he left me hanging...
Since I understand that a github issue tracker is no place to discuss this further, I added my mailaddress to my profile, so please feel free to contact me!

jremen, 2022-08-17:
@fcollingwood it would really be great if this project would not die because of not being able to continue due to ctb encryption.
Please, provide PR for ctb encrypted support.

zackfuchtel, 2023-07-20:
> @fcollingwood it would really be great if this project would not die because of not being able to continue due to ctb encryption. Please, provide PR for ctb encrypted support.

Full quote, me too

amd989, 2026-11-11:
Fixed in [Mariner 1.0.0](https://github.com/mariner3d/mariner/releases/tag/v1.0.0): it reads encrypted `.ctb` files, which is what ChiTuBox 1.9 and later write.

If your files still fail on 1.0.0, comment here with the slicer version and the log and I'll reopen.

closed 2026-11-11 by amd989