luizribeiro/mariner #479: config.toml not working

fcollingwood, 2021-11-04:
**Description**
I am attempting to install on a Orange Pi One

I have placed the following config.toml in /etc/mariner:

 `# mariner configuration file
# =============================================================================
# This file holds the configuration of mariner. It can be placed on either of
# the following paths:
#   * ./config.toml (the running directory)
#   * ~/.mariner/config.toml (home directory)
#   * /etc/mariner/config.toml
#
# The config is formatted in the TOML format, see https://toml.io/en/ for
# details on how TOML formats work.

# The directory to which files should be uploaded. This is usually the same
# directory that you are sharing to the printer through the USB Gadget
files_directory = "/mnt/usb_share"

[printer]
# The name which will be displayed for the printer in the UI. By default no
# printer name is displayed. Uncomment this line and set it to whatever you
# want if you want to show a name on the print status. Useful for people with
# multiple printers running multiple instances of mariner.
display_name = "Mighty 4K 1"

# Serial port settings. Most of the time the default values are fine. This
# is generally only helpful if you aren't running mariner on a Raspberry Pi
# or if you are running with a printer that uses a different baud rate.
serial_port = "/dev/ttyS1"
baudrate = 115200

[http]
# Hostname or IP address (string) on which to listen. Default is 0.0.0.0, which
# means "all IP addresses on this host". Note that listening on 0.0.0.0 assumes
# that your mariner instance is running on a trusted network.
host = "0.0.0.0"
# TCP port on which to listen
port = 5000

[cache]
# Directory in which cached information such as file metadata and thumbnails
# will be stored
directory = "/tmp/mariner/"
`

However, it still defaults to trying to connect to the printer on /dev/serial0:

`● mariner3d.service - mariner3d Printer Controller
   Loaded: loaded (/lib/systemd/system/mariner3d.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2021-11-04 19:33:21 NZDT; 6min ago
 Main PID: 1764 (mariner)
    Tasks: 6 (limit: 905)
   Memory: 18.9M
   CGroup: /system.slice/mariner3d.service
           └─1764 /opt/venvs/mariner3d/bin/python /usr/bin/mariner

Nov 04 19:38:52 orangepione mariner[1764]:     return self.view_functions[rule.endpoint](**req.view_args)
Nov 04 19:38:52 orangepione mariner[1764]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/server/api.py", line 45, in print_status
Nov 04 19:38:52 orangepione mariner[1764]:     with ElegooMars() as elegoo_mars:
Nov 04 19:38:52 orangepione mariner[1764]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/mars.py", line 50, in __enter__
Nov 04 19:38:52 orangepione mariner[1764]:     self.open()
Nov 04 19:38:52 orangepione mariner[1764]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/mariner/mars.py", line 44, in open
Nov 04 19:38:52 orangepione mariner[1764]:     self._serial_port.open()
Nov 04 19:38:52 orangepione mariner[1764]:   File "/opt/venvs/mariner3d/lib/python3.7/site-packages/serial/serialposix.py", line 325, in open
Nov 04 19:38:52 orangepione mariner[1764]:     raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
Nov 04 19:38:52 orangepione mariner[1764]: serial.serialutil.SerialException: [Errno 2] could not open port /dev/serial0: [Errno 2] No such file or directory: '/dev/serial0'
`

I have tested to make sure it can actually work by softlinking /dev/serial0 to /dev/ttyS1 and when I do this it can connect to the printer. However, this isn't the best solution and won't survive a reboot anyway.

luizribeiro, 2021-11-06:
Which build are you using? There isn't a release that supports `config.toml` yet. So unless you are using a build from CI, it won't work

fcollingwood, 2021-11-07:
Not sure what you mean about the build from CI. What's CI?

I just followed the installation instructions, so it's whichever build is in the https://ppa.l9o.dev/raspbian repo.

I have used udev to create a permanent link in /dev for serial0, but it would be good to be able to customise it. Also, I notice that no printer name is shown the web console for the repo build

luizribeiro, 2021-11-07:
CI stands for Continuous Integration. Every commit to the repo runs all the tests and builds a `.deb` against that commit.

So you can download that `.deb` from the artifacts of that GitHub Action run. The current one is here: https://github.com/luizribeiro/mariner/actions/runs/1018824027

luizribeiro, 2021-11-07:
It seems like this wasn't really an issue, but rather just running the wrong version. Maybe I should just release a 0.3 version even though we don't have all the features I'd like for that version :)

Going to close this issue, but feel free to reopen if your problem isn't solved.

closed 2021-11-07 by luizribeiro

fcollingwood, 2021-11-07:
Ah! Thanks for that. Unfortunately that .deb has expired, so I can't download and install. I've got it working fine with the udev hack, so I'll wait until 0.3. Looking forward to more goodness in the package. I'll pop in a few feature requests

SaKiEQ, 2022-01-09:
Ha - that explains why I could not get .toml to work ! Needed to change the default port away from 5000 so it could co-exist on a OctoPrint setup... Ended up changing the default port in the code instead...