luizribeiro/mariner #481: ModuleNotFoundError: No module named 'encodings'

jangrewe, 2021-11-12:
**Description**
Mariner can't be started on DietPi

**Reproduction Steps**
Steps to reproduce the behavior:
1. Install DietPi on SD card, boot and do minimal setup to get a running system.
2. Follow setup instructions to install mariner package from PPA
3. (optional) Install `dbus` to make systemd behave properly

**Expected Behavior**
Mariner starts, preferably from the systemd unit, but manual start via `sudo -u mariner mariner` is acceptable.

**Screenshots / Logs**
```
dietpi@marsbox:~$ sudo -u mariner mariner
Could not find platform independent libraries <prefix>
Could not find platform dependent libraries <exec_prefix>
Consider setting $PYTHONHOME to <prefix>[:<exec_prefix>]
Fatal Python error: initfsencoding: Unable to get the locale encoding
ModuleNotFoundError: No module named 'encodings'

Current thread 0xb6fc6a00 (most recent call first):
Aborted
```

**Details:**
 - Mariner Version: 0.2.0-1
 - Printer Model: Elegoo Mars
 - Raspberry Pi Model: RPi Zero W

**Additional context**
Add any other context about the problem here.

jangrewe, 2021-11-13:
I guess this is the problem:
```
dietpi@marsbox:~$ ls -l /opt/venvs/mariner3d/lib/python3.7/encodings
lrwxrwxrwx 1 root root 28 Apr 17  2021 /opt/venvs/mariner3d/lib/python3.7/encodings -> /usr/lib/python3.7/encodings
```
But DietPi (based on Rasbian/Debian 11) has Python 3.9, so the correct target for the symlink would be `/usr/lib/python3.9/encodings`.
Trying the stupid thing and symlinking `/usr/lib/python3.7` to `/usr/lib/python3.9` helps with a couple of errors, but not enough to get it to work.
I think the big issue is that you're using a packaged venv that makes incorrect assumptions about the environment it gets installed to. Maybe it would be better do do a `pip3 install .` from sources, when the package gets installed? That would at least ensure the correct environment is used.

dmpanch, 2021-11-14:
Seems like postinstall script set Python 3.7 instead of Python 3.9 available in the system

jangrewe, 2021-11-14:
Thanks @dmpanch, i've now rebuilt everything against Python 3.9 and got it to work, at least theoretically. Your approach is probably a lot faster, so i'll give that a shot for the next time.

But i guess it would be nicer if the package would be built for the current Python version, i guess.

Anyways, now i'm running into a problem similar to yours (which doesn't look like it's python related, though):
**The printer returned an unexpected response: 'ok N:20\r\n'**
```
Traceback (most recent call last):
  File "/opt/venvs/mariner3d/lib/python3.9/site-packages/flask/app.py", line 1513, in full_dispatch_request
    rv = self.dispatch_request()
  File "/opt/venvs/mariner3d/lib/python3.9/site-packages/flask/app.py", line 1499, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**req.view_args)
  File "/opt/mariner/mariner/server/api.py", line 54, in print_status
    selected_file = retry(
  File "/opt/mariner/mariner/server/utils.py", line 54, in retry
    return func()
  File "/opt/mariner/mariner/printer.py", line 103, in get_selected_file
    self._extract_response_with_regex("ok '([^']+)'\r\n", data).group(1)
  File "/opt/mariner/mariner/printer.py", line 40, in _extract_response_with_regex
    raise UnexpectedPrinterResponse(data)
mariner.exceptions.UnexpectedPrinterResponse: ok N:20

```

dmpanch, 2021-11-17:
@jangrewe totally agree with you about the Python version. Btw, today I found [PPA with the different Python versions](https://launchpad.net/~deadsnakes/+archive/ubuntu/ppa), it would have been useful a few days ago.

Moba160, 2022-01-02:
Thank you dmpanch, you saved my day.

There is only one slight mistake in your ln command, the parameters need to be exchanged; the correct command is
`sudo ln -s  /usr/bin/python3.9 python3.7`

antwal, 2022-01-07:
Problem solved, check here #493 for install python 3.7.4

hiddentapir, 2022-01-21:
I got around this by starting with a legacy raspbian image. The Debian Buster based image released Dec 2nd 2021 comes with python 3.7 instead of 3.9, and Mariner installs & runs with no issues.

kane-thornwyrd, 2022-02-20:
‼️https://github.com/luizribeiro/mariner/issues/481#issuecomment-968325624 is harmful because you are basically mixing two versions of python hoping for the best, maybe it worked, fine, but you may have screwed the installation of any other softwares relying on Python 3.9 you may want to install later.

amd989, 2026-11-11:
Closing: this was about installing on Raspbian buster with Python 3.7 from the old PPA. [Mariner 1.0.0](https://github.com/mariner3d/mariner/releases/tag/v1.0.0) ships for Raspberry Pi OS bookworm and trixie with its own Python, so these steps no longer apply. Install instructions: https://mariner.readthedocs.io/en/latest/software-setup.html

If this still happens on 1.0.0, comment here and I'll reopen.

closed 2026-11-11 by amd989