luizribeiro/mariner #482: mariner3d wont give me an active signal

jivry5, 2021-11-13:
**Description**
I tried manually installing mariner on my pi zero twice and once with Mariner Autoinstaller and I get the same issue where when I check the status of mariner3d it says Active: failed

pi@mariner:~ $ sudo systemctl status mariner3d
● mariner3d.service - mariner3d Printer Controller
     Loaded: loaded (/lib/systemd/system/mariner3d.service; enabled; vendor preset: enabled)
     Active: failed (Result: signal) since Sat 2021-11-13 15:09:52 GMT; 2min 49s ago
    Process: 262 ExecStart=/usr/bin/mariner (code=killed, signal=ABRT)
   Main PID: 262 (code=killed, signal=ABRT)
        CPU: 132ms

Nov 13 15:09:50 mariner systemd[1]: Started mariner3d Printer Controller.
Nov 13 15:09:51 mariner mariner[262]: Could not find platform independent libraries <prefix>
Nov 13 15:09:51 mariner mariner[262]: Could not find platform dependent libraries <exec_prefix>
Nov 13 15:09:51 mariner mariner[262]: Consider setting $PYTHONHOME to <prefix>[:<exec_prefix>]
Nov 13 15:09:52 mariner mariner[262]: Fatal Python error: initfsencoding: Unable to get the locale encoding
Nov 13 15:09:52 mariner mariner[262]: ModuleNotFoundError: No module named 'encodings'
Nov 13 15:09:52 mariner mariner[262]: Current thread 0xb6f9aa00 (most recent call first):
Nov 13 15:09:52 mariner systemd[1]: mariner3d.service: Main process exited, code=killed, status=6/ABRT
Nov 13 15:09:52 mariner systemd[1]: mariner3d.service: Failed with result 'signal'.
pi@mariner:~ $

**Screenshots / Logs**
pi@mariner:~ $ sudo journalctl -u mariner3d
-- Journal begins at Sat 2021-10-30 12:29:42 BST, ends at Sat 2021-11-13 15:18:06 GMT. --
Nov 13 14:11:45 mariner systemd[1]: Started mariner3d Printer Controller.
Nov 13 14:11:45 mariner mariner[5592]: Could not find platform independent libraries <prefix>
Nov 13 14:11:45 mariner mariner[5592]: Could not find platform dependent libraries <exec_prefix>
Nov 13 14:11:45 mariner mariner[5592]: Consider setting $PYTHONHOME to <prefix>[:<exec_prefix>]
Nov 13 14:11:45 mariner mariner[5592]: Fatal Python error: initfsencoding: Unable to get the locale encoding
Nov 13 14:11:45 mariner mariner[5592]: ModuleNotFoundError: No module named 'encodings'
Nov 13 14:11:45 mariner mariner[5592]: Current thread 0xb6f3aa00 (most recent call first):
Nov 13 14:11:45 mariner systemd[1]: mariner3d.service: Main process exited, code=killed, status=6/ABRT
Nov 13 14:11:45 mariner systemd[1]: mariner3d.service: Failed with result 'signal'.
-- Boot 729d4ba386984a6484d264250b752813 --
Nov 13 14:28:41 mariner systemd[1]: Started mariner3d Printer Controller.
Nov 13 14:28:42 mariner mariner[254]: Could not find platform independent libraries <prefix>
Nov 13 14:28:42 mariner mariner[254]: Could not find platform dependent libraries <exec_prefix>
Nov 13 14:28:42 mariner mariner[254]: Consider setting $PYTHONHOME to <prefix>[:<exec_prefix>]
Nov 13 14:28:43 mariner mariner[254]: Fatal Python error: initfsencoding: Unable to get the locale encoding
Nov 13 14:28:43 mariner mariner[254]: ModuleNotFoundError: No module named 'encodings'
Nov 13 14:28:43 mariner mariner[254]: Current thread 0xb6f1ca00 (most recent call first):
Nov 13 14:28:43 mariner systemd[1]: mariner3d.service: Main process exited, code=killed, status=6/ABRT
Nov 13 14:28:43 mariner systemd[1]: mariner3d.service: Failed with result 'signal'.
-- Boot f08210c648b0484c841fe28cc448df24 --
Nov 13 14:35:48 mariner systemd[1]: Started mariner3d Printer Controller.
Nov 13 14:35:49 mariner mariner[258]: Could not find platform independent libraries <prefix>
Nov 13 14:35:49 mariner mariner[258]: Could not find platform dependent libraries <exec_prefix>
Nov 13 14:35:49 mariner mariner[258]: Consider setting $PYTHONHOME to <prefix>[:<exec_prefix>]
Nov 13 14:35:49 mariner mariner[258]: Fatal Python error: initfsencoding: Unable to get the locale encoding
Nov 13 14:35:49 mariner mariner[258]: ModuleNotFoundError: No module named 'encodings'
Nov 13 14:35:49 mariner mariner[258]: Current thread 0xb6f8ca00 (most recent call first):
Nov 13 14:35:50 mariner systemd[1]: mariner3d.service: Main process exited, code=killed, status=6/ABRT
Nov 13 14:35:50 mariner systemd[1]: mariner3d.service: Failed with result 'signal'.
-- Boot 44cc669f82814245b68c05fb97b8db9e --
Nov 13 14:49:06 mariner systemd[1]: Started mariner3d Printer Controller.
Nov 13 14:49:07 mariner mariner[261]: Could not find platform independent libraries <prefix>
Nov 13 14:49:07 mariner mariner[261]: Could not find platform dependent libraries <exec_prefix>
Nov 13 14:49:07 mariner mariner[261]: Consider setting $PYTHONHOME to <prefix>[:<exec_prefix>]
Nov 13 14:49:08 mariner mariner[261]: Fatal Python error: initfsencoding: Unable to get the locale encoding
Nov 13 14:49:08 mariner mariner[261]: ModuleNotFoundError: No module named 'encodings'
Nov 13 14:49:08 mariner mariner[261]: Current thread 0xb6f87a00 (most recent call first):
Nov 13 14:49:08 mariner systemd[1]: mariner3d.service: Main process exited, code=killed, status=6/ABRT
Nov 13 14:49:08 mariner systemd[1]: mariner3d.service: Failed with result 'signal'.
-- Boot 8a8fa329971c40b9a85aa4c05c6a9ba5 --
Nov 13 15:09:50 mariner systemd[1]: Started mariner3d Printer Controller.
Nov 13 15:09:51 mariner mariner[262]: Could not find platform independent libraries <prefix>
Nov 13 15:09:51 mariner mariner[262]: Could not find platform dependent libraries <exec_prefix>
Nov 13 15:09:51 mariner mariner[262]: Consider setting $PYTHONHOME to <prefix>[:<exec_prefix>]
Nov 13 15:09:52 mariner mariner[262]: Fatal Python error: initfsencoding: Unable to get the locale encoding
Nov 13 15:09:52 mariner mariner[262]: ModuleNotFoundError: No module named 'encodings'
Nov 13 15:09:52 mariner mariner[262]: Current thread 0xb6f9aa00 (most recent call first):
Nov 13 15:09:52 mariner systemd[1]: mariner3d.service: Main process exited, code=killed, status=6/ABRT
Nov 13 15:09:52 mariner systemd[1]: mariner3d.service: Failed with result 'signal'.
pi@mariner:~ $

**Details:**
 - Mariner Version: 0.2.0-1 
 - Printer Model: Elegoo Saturn
 - Raspberry Pi Model:  RPi Zero W

thank you so much for your time

Tim2309, 2021-11-23:
I got the same problem with a RPi Zero W 2
The network drive work, but i can't run Mariner

JMAN7102, 2021-12-03:
I'm also getting similar problems with the exact same error logs. I can see the network share but cannot actually use the webGUI at all. Has anybody gotten a workaround/solution for remotely starting prints if I can't hit the webpage?

pi@mariner:~ $ sudo systemctl status mariner3d
● mariner3d.service - mariner3d Printer Controller
     Loaded: loaded (/lib/systemd/system/mariner3d.service; enabled; vendor preset: enabled)
     Active: failed (Result: signal) since Fri 2021-12-03 01:53:50 GMT; 7min ago
    Process: 337 ExecStart=/usr/bin/mariner (code=killed, signal=ABRT)
   Main PID: 337 (code=killed, signal=ABRT)
        CPU: 78ms

Dec 03 01:53:50 mariner systemd[1]: Started mariner3d Printer Controller.
Dec 03 01:53:50 mariner mariner[337]: Could not find platform independent libraries <prefix>
Dec 03 01:53:50 mariner mariner[337]: Could not find platform dependent libraries <exec_prefix>
Dec 03 01:53:50 mariner mariner[337]: Consider setting $PYTHONHOME to <prefix>[:<exec_prefix>]
Dec 03 01:53:50 mariner mariner[337]: Fatal Python error: initfsencoding: Unable to get the locale encoding
Dec 03 01:53:50 mariner mariner[337]: ModuleNotFoundError: No module named 'encodings'
Dec 03 01:53:50 mariner mariner[337]: Current thread 0x76f43f40 (most recent call first):
Dec 03 01:53:50 mariner systemd[1]: mariner3d.service: Main process exited, code=killed, status=6/ABRT
Dec 03 01:53:50 mariner systemd[1]: mariner3d.service: Failed with result 'signal'.

pi@mariner:~ $ sudo journalctl -u mariner3d
-- Journal begins at Sat 2021-10-30 12:29:41 BST, ends at Fri 2021-12-03 02:01:44 GMT. --
Dec 02 07:07:35 mariner systemd[1]: Started mariner3d Printer Controller.
Dec 02 07:07:35 mariner mariner[2131]: Could not find platform independent libraries <prefix>
Dec 02 07:07:35 mariner mariner[2131]: Could not find platform dependent libraries <exec_prefix>
Dec 02 07:07:35 mariner mariner[2131]: Consider setting $PYTHONHOME to <prefix>[:<exec_prefix>]
Dec 02 07:07:35 mariner mariner[2131]: Fatal Python error: initfsencoding: Unable to get the locale encoding
Dec 02 07:07:35 mariner mariner[2131]: ModuleNotFoundError: No module named 'encodings'
Dec 02 07:07:35 mariner mariner[2131]: Current thread 0x76f3df40 (most recent call first):
Dec 02 07:07:35 mariner systemd[1]: mariner3d.service: Main process exited, code=killed, status=6/ABRT
Dec 02 07:07:35 mariner systemd[1]: mariner3d.service: Failed with result 'signal'.
-- Boot 73433fe9b23347f4a403e4d8dfcd7b15 --
Dec 02 07:07:56 mariner systemd[1]: Started mariner3d Printer Controller.
Dec 02 07:07:56 mariner mariner[335]: Could not find platform independent libraries <prefix>
Dec 02 07:07:56 mariner mariner[335]: Could not find platform dependent libraries <exec_prefix>
Dec 02 07:07:56 mariner mariner[335]: Consider setting $PYTHONHOME to <prefix>[:<exec_prefix>]
Dec 02 07:07:56 mariner mariner[335]: Fatal Python error: initfsencoding: Unable to get the locale encoding
Dec 02 07:07:56 mariner mariner[335]: ModuleNotFoundError: No module named 'encodings'
Dec 02 07:07:56 mariner mariner[335]: Current thread 0x76f00f40 (most recent call first):
Dec 02 07:07:56 mariner systemd[1]: mariner3d.service: Main process exited, code=killed, status=6/ABRT
Dec 02 07:07:56 mariner systemd[1]: mariner3d.service: Failed with result 'signal'.
Dec 02 07:17:32 mariner systemd[1]: Started mariner3d Printer Controller.
Dec 02 07:17:32 mariner mariner[714]: Could not find platform independent libraries <prefix>
Dec 02 07:17:32 mariner mariner[714]: Could not find platform dependent libraries <exec_prefix>
Dec 02 07:17:32 mariner mariner[714]: Consider setting $PYTHONHOME to <prefix>[:<exec_prefix>]
Dec 02 07:17:32 mariner mariner[714]: Fatal Python error: initfsencoding: Unable to get the locale encoding
Dec 02 07:17:32 mariner mariner[714]: ModuleNotFoundError: No module named 'encodings'
Dec 02 07:17:32 mariner mariner[714]: Current thread 0x76fcaf40 (most recent call first):
Dec 02 07:17:32 mariner systemd[1]: mariner3d.service: Main process exited, code=killed, status=6/ABRT
Dec 02 07:17:32 mariner systemd[1]: mariner3d.service: Failed with result 'signal'.
-- Boot debafbc1a38b40a3ba6ea6b92d4cc44a --
Dec 02 07:18:15 mariner systemd[1]: Started mariner3d Printer Controller.
Dec 02 07:18:15 mariner mariner[330]: Could not find platform independent libraries <prefix>
Dec 02 07:18:15 mariner mariner[330]: Could not find platform dependent libraries <exec_prefix>
Dec 02 07:18:15 mariner mariner[330]: Consider setting $PYTHONHOME to <prefix>[:<exec_prefix>]
Dec 02 07:18:15 mariner mariner[330]: Fatal Python error: initfsencoding: Unable to get the locale encoding
Dec 02 07:18:15 mariner mariner[330]: ModuleNotFoundError: No module named 'encodings'
Dec 02 07:18:15 mariner mariner[330]: Current thread 0x76f9df40 (most recent call first):
Dec 02 07:18:15 mariner systemd[1]: mariner3d.service: Main process exited, code=killed, status=6/ABRT
Dec 02 07:18:15 mariner systemd[1]: mariner3d.service: Failed with result 'signal'.
Dec 02 07:38:53 mariner systemd[1]: Started mariner3d Printer Controller.
Dec 02 07:38:53 mariner mariner[1048]: Could not find platform independent libraries <prefix>
Dec 02 07:38:53 mariner mariner[1048]: Could not find platform dependent libraries <exec_prefix>
Dec 02 07:38:53 mariner mariner[1048]: Consider setting $PYTHONHOME to <prefix>[:<exec_prefix>]
Dec 02 07:38:53 mariner mariner[1048]: Fatal Python error: initfsencoding: Unable to get the locale encoding
[... cut ...]
Dec 03 01:53:50 mariner mariner[337]: Consider setting $PYTHONHOME to <prefix>[:<exec_prefix>]
Dec 03 01:53:50 mariner mariner[337]: Fatal Python error: initfsencoding: Unable to get the locale encoding
Dec 03 01:53:50 mariner mariner[337]: ModuleNotFoundError: No module named 'encodings'
Dec 03 01:53:50 mariner mariner[337]: Current thread 0x76f43f40 (most recent call first):
Dec 03 01:53:50 mariner systemd[1]: mariner3d.service: Main process exited, code=killed, status=6/ABRT
Dec 03 01:53:50 mariner systemd[1]: mariner3d.service: Failed with result 'signal'.

JMAN7102, 2021-12-03:
@jivry5 I found the solution on another issue if you were still having the problem: https://github.com/luizribeiro/mariner/issues/481#issuecomment-968325624

It looks like the issue is the python versions aren't consistent. that post has the commands to fix it, I just used them to set my installation up.

jackslao, 2021-12-04:
Hi,

I just tried the above method and it looks like I can get through the web interface now. However, seems like there is still no signal. Here's what I see after I type sudo systemctl status mariner3d

● mariner3d.service - mariner3d Printer Controller
     Loaded: loaded (/lib/systemd/system/mariner3d.service; enabled; vendor preset: enabled)
     Active: active (running) since Fri 2021-12-03 16:20:01 PST; 2min 22s ago
   Main PID: 358 (mariner)
      Tasks: 6 (limit: 409)
        CPU: 9.247s
     CGroup: /system.slice/mariner3d.service
             └─358 /opt/venvs/mariner3d/bin/python /usr/bin/mariner

Dec 03 16:21:46 raspberrypi mariner[358]:     return self.view_functions[rule.endpoint](**req.view_args)
Dec 03 16:21:46 raspberrypi mariner[358]:   File "/opt/venvs/mariner3d/lib/python3.9/site-packages/mariner/server/api.p>
Dec 03 16:21:46 raspberrypi mariner[358]:     with ElegooMars() as elegoo_mars:
Dec 03 16:21:46 raspberrypi mariner[358]:   File "/opt/venvs/mariner3d/lib/python3.9/site-packages/mariner/mars.py", li>
Dec 03 16:21:46 raspberrypi mariner[358]:     self.open()
Dec 03 16:21:46 raspberrypi mariner[358]:   File "/opt/venvs/mariner3d/lib/python3.9/site-packages/mariner/mars.py", li>
Dec 03 16:21:46 raspberrypi mariner[358]:     self._serial_port.open()
Dec 03 16:21:46 raspberrypi mariner[358]:   File "/opt/venvs/mariner3d/lib/python3.9/site-packages/serial/serialposix.p>
Dec 03 16:21:46 raspberrypi mariner[358]:     raise SerialException(msg.errno, "could not open port {}: {}".format(self>
Dec 03 16:21:46 raspberrypi mariner[358]: serial.serialutil.SerialException: [Errno 13] could not open port /dev/serial>

My web interface gives me an error of:
"Something went wrong, the server replied with a 500 HTTP status code"

And the printer status is in a infinite cycle of refresh wheel.

I too am using Pi Zero 2.

JMAN7102, 2021-12-04:
@jackslao Are you trying to print using a USB and just monitoring the printer with Mariner? Similar to this thread:https://github.com/luizribeiro/mariner/issues/454#issue-971846211 ?

Can you paste out the result from

`sudo journalctl -u mariner3d`

jackslao, 2021-12-04:
No I am trying to get full function of the mariner3d. Like other folks, I am able to access and import files fine. It's just that it seems like Pi is not seeing/reading the signal of the printer...

Here's the last group of the result from 

`sudo journalctl -u mariner3d`

Dec 03 16:28:48 raspberrypi mariner[358]: ERROR:waitress:Exception while serving /api/print_status
Dec 03 16:28:48 raspberrypi mariner[358]: Traceback (most recent call last):
Dec 03 16:28:48 raspberrypi mariner[358]:   File "/opt/venvs/mariner3d/lib/python3.9/site-packages/serial/serialposix.py", line 322, in open
Dec 03 16:28:48 raspberrypi mariner[358]:     self.fd = os.open(self.portstr, os.O_RDWR | os.O_NOCTTY | os.O_NONBLOCK)
Dec 03 16:28:48 raspberrypi mariner[358]: PermissionError: [Errno 13] Permission denied: '/dev/serial0'
Dec 03 16:28:48 raspberrypi mariner[358]: During handling of the above exception, another exception occurred:
Dec 03 16:28:48 raspberrypi mariner[358]: Traceback (most recent call last):
Dec 03 16:28:48 raspberrypi mariner[358]:   File "/opt/venvs/mariner3d/lib/python3.9/site-packages/waitress/channel.py", line 397, in service
Dec 03 16:28:48 raspberrypi mariner[358]:     task.service()
Dec 03 16:28:48 raspberrypi mariner[358]:   File "/opt/venvs/mariner3d/lib/python3.9/site-packages/waitress/task.py", line 168, in service
Dec 03 16:28:48 raspberrypi mariner[358]:     self.execute()
Dec 03 16:28:48 raspberrypi mariner[358]:   File "/opt/venvs/mariner3d/lib/python3.9/site-packages/waitress/task.py", line 434, in execute
Dec 03 16:28:48 raspberrypi mariner[358]:     app_iter = self.channel.server.application(environ, start_response)
Dec 03 16:28:48 raspberrypi mariner[358]:   File "/opt/venvs/mariner3d/lib/python3.9/site-packages/flask/app.py", line 2464, in __call__
Dec 03 16:28:48 raspberrypi mariner[358]:     return self.wsgi_app(environ, start_response)
Dec 03 16:28:48 raspberrypi mariner[358]:   File "/opt/venvs/mariner3d/lib/python3.9/site-packages/whitenoise/base.py", line 85, in __call__
Dec 03 16:28:48 raspberrypi mariner[358]:     return self.application(environ, start_response)
Dec 03 16:28:48 raspberrypi mariner[358]:   File "/opt/venvs/mariner3d/lib/python3.9/site-packages/flask/app.py", line 2450, in wsgi_app
Dec 03 16:28:48 raspberrypi mariner[358]:     response = self.handle_exception(e)
Dec 03 16:28:48 raspberrypi mariner[358]:   File "/opt/venvs/mariner3d/lib/python3.9/site-packages/flask/app.py", line 1867, in handle_exception
Dec 03 16:28:48 raspberrypi mariner[358]:     reraise(exc_type, exc_value, tb)
Dec 03 16:28:48 raspberrypi mariner[358]:   File "/opt/venvs/mariner3d/lib/python3.9/site-packages/flask/_compat.py", line 39, in reraise
Dec 03 16:28:48 raspberrypi mariner[358]:     raise value
Dec 03 16:28:48 raspberrypi mariner[358]:   File "/opt/venvs/mariner3d/lib/python3.9/site-packages/flask/app.py", line 2447, in wsgi_app
Dec 03 16:28:48 raspberrypi mariner[358]:     response = self.full_dispatch_request()
Dec 03 16:28:48 raspberrypi mariner[358]:   File "/opt/venvs/mariner3d/lib/python3.9/site-packages/flask/app.py", line 1952, in full_dispatch_request
Dec 03 16:28:48 raspberrypi mariner[358]:     rv = self.handle_user_exception(e)
Dec 03 16:28:48 raspberrypi mariner[358]:   File "/opt/venvs/mariner3d/lib/python3.9/site-packages/flask/app.py", line 1821, in handle_user_exception
Dec 03 16:28:48 raspberrypi mariner[358]:     reraise(exc_type, exc_value, tb)
Dec 03 16:28:48 raspberrypi mariner[358]:   File "/opt/venvs/mariner3d/lib/python3.9/site-packages/flask/_compat.py", line 39, in reraise
Dec 03 16:28:48 raspberrypi mariner[358]:     raise value
Dec 03 16:28:48 raspberrypi mariner[358]:   File "/opt/venvs/mariner3d/lib/python3.9/site-packages/flask/app.py", line 1950, in full_dispatch_request
Dec 03 16:28:48 raspberrypi mariner[358]:     rv = self.dispatch_request()
Dec 03 16:28:48 raspberrypi mariner[358]:   File "/opt/venvs/mariner3d/lib/python3.9/site-packages/flask/app.py", line 1936, in dispatch_request
Dec 03 16:28:48 raspberrypi mariner[358]:     return self.view_functions[rule.endpoint](**req.view_args)
Dec 03 16:28:48 raspberrypi mariner[358]:   File "/opt/venvs/mariner3d/lib/python3.9/site-packages/mariner/server/api.py", line 45, in print_status
Dec 03 16:28:48 raspberrypi mariner[358]:     with ElegooMars() as elegoo_mars:
Dec 03 16:28:48 raspberrypi mariner[358]:   File "/opt/venvs/mariner3d/lib/python3.9/site-packages/mariner/mars.py", line 50, in __enter__
Dec 03 16:28:48 raspberrypi mariner[358]:     self.open()
Dec 03 16:28:48 raspberrypi mariner[358]:   File "/opt/venvs/mariner3d/lib/python3.9/site-packages/mariner/mars.py", line 44, in open
Dec 03 16:28:48 raspberrypi mariner[358]:     self._serial_port.open()
Dec 03 16:28:48 raspberrypi mariner[358]:   File "/opt/venvs/mariner3d/lib/python3.9/site-packages/serial/serialposix.py", line 325, in open
Dec 03 16:28:48 raspberrypi mariner[358]:     raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
Dec 03 16:28:48 raspberrypi mariner[358]: serial.serialutil.SerialException: [Errno 13] could not open port /dev/serial0: [Errno 13] Permission denied: '/dev/serial0'

JMAN7102, 2021-12-04:
It looks like there is an issue with it actually opening the communication on the serial port. When you run `groups` when logged in as pi is the group dialout listed? If not, try this: `sudo usermod -a -G dialout pi`

Normally I think this is done automatically but you might want to check that out.

jackslao, 2021-12-04:
Hi,

The `groups` have dialout in it.

'groups
pi adm dialout cdrom sudo audio video plugdev games users input netdev gpio i2c spi'

But the same problem still occured...

Here's the journal log after I put in 'sudo usermod -a -G dialout pi'

Dec 04 13:51:48 raspberrypi mariner[349]: ERROR:waitress:Exception while serving /api/print_status
Dec 04 13:51:48 raspberrypi mariner[349]: Traceback (most recent call last):
Dec 04 13:51:48 raspberrypi mariner[349]:   File "/opt/venvs/mariner3d/lib/python3.9/site-packages/serial/serialposix.py", line 322, in open
Dec 04 13:51:48 raspberrypi mariner[349]:     self.fd = os.open(self.portstr, os.O_RDWR | os.O_NOCTTY | os.O_NONBLOCK)
Dec 04 13:51:48 raspberrypi mariner[349]: PermissionError: [Errno 13] Permission denied: '/dev/serial0'
Dec 04 13:51:48 raspberrypi mariner[349]: During handling of the above exception, another exception occurred:
Dec 04 13:51:48 raspberrypi mariner[349]: Traceback (most recent call last):
Dec 04 13:51:48 raspberrypi mariner[349]:   File "/opt/venvs/mariner3d/lib/python3.9/site-packages/waitress/channel.py", line 397, in service
Dec 04 13:51:48 raspberrypi mariner[349]:     task.service()
Dec 04 13:51:48 raspberrypi mariner[349]:   File "/opt/venvs/mariner3d/lib/python3.9/site-packages/waitress/task.py", line 168, in service
Dec 04 13:51:48 raspberrypi mariner[349]:     self.execute()
Dec 04 13:51:48 raspberrypi mariner[349]:   File "/opt/venvs/mariner3d/lib/python3.9/site-packages/waitress/task.py", line 434, in execute
Dec 04 13:51:48 raspberrypi mariner[349]:     app_iter = self.channel.server.application(environ, start_response)
Dec 04 13:51:48 raspberrypi mariner[349]:   File "/opt/venvs/mariner3d/lib/python3.9/site-packages/flask/app.py", line 2464, in __call__
Dec 04 13:51:48 raspberrypi mariner[349]:     return self.wsgi_app(environ, start_response)
Dec 04 13:51:48 raspberrypi mariner[349]:   File "/opt/venvs/mariner3d/lib/python3.9/site-packages/whitenoise/base.py", line 85, in __call__
Dec 04 13:51:48 raspberrypi mariner[349]:     return self.application(environ, start_response)
Dec 04 13:51:48 raspberrypi mariner[349]:   File "/opt/venvs/mariner3d/lib/python3.9/site-packages/flask/app.py", line 2450, in wsgi_app
Dec 04 13:51:48 raspberrypi mariner[349]:     response = self.handle_exception(e)
Dec 04 13:51:48 raspberrypi mariner[349]:   File "/opt/venvs/mariner3d/lib/python3.9/site-packages/flask/app.py", line 1867, in handle_exception
Dec 04 13:51:48 raspberrypi mariner[349]:     reraise(exc_type, exc_value, tb)
Dec 04 13:51:48 raspberrypi mariner[349]:   File "/opt/venvs/mariner3d/lib/python3.9/site-packages/flask/_compat.py", line 39, in reraise
Dec 04 13:51:48 raspberrypi mariner[349]:     raise value
Dec 04 13:51:48 raspberrypi mariner[349]:   File "/opt/venvs/mariner3d/lib/python3.9/site-packages/flask/app.py", line 2447, in wsgi_app
Dec 04 13:51:48 raspberrypi mariner[349]:     response = self.full_dispatch_request()
Dec 04 13:51:48 raspberrypi mariner[349]:   File "/opt/venvs/mariner3d/lib/python3.9/site-packages/flask/app.py", line 1952, in full_dispatch_request
Dec 04 13:51:48 raspberrypi mariner[349]:     rv = self.handle_user_exception(e)
Dec 04 13:51:48 raspberrypi mariner[349]:   File "/opt/venvs/mariner3d/lib/python3.9/site-packages/flask/app.py", line 1821, in handle_user_exception
Dec 04 13:51:48 raspberrypi mariner[349]:     reraise(exc_type, exc_value, tb)
Dec 04 13:51:48 raspberrypi mariner[349]:   File "/opt/venvs/mariner3d/lib/python3.9/site-packages/flask/_compat.py", line 39, in reraise
Dec 04 13:51:48 raspberrypi mariner[349]:     raise value
Dec 04 13:51:48 raspberrypi mariner[349]:   File "/opt/venvs/mariner3d/lib/python3.9/site-packages/flask/app.py", line 1950, in full_dispatch_request
Dec 04 13:51:48 raspberrypi mariner[349]:     rv = self.dispatch_request()
Dec 04 13:51:48 raspberrypi mariner[349]:   File "/opt/venvs/mariner3d/lib/python3.9/site-packages/flask/app.py", line 1936, in dispatch_request
Dec 04 13:51:48 raspberrypi mariner[349]:     return self.view_functions[rule.endpoint](**req.view_args)
Dec 04 13:51:48 raspberrypi mariner[349]:   File "/opt/venvs/mariner3d/lib/python3.9/site-packages/mariner/server/api.py", line 45, in print_status
Dec 04 13:51:48 raspberrypi mariner[349]:     with ElegooMars() as elegoo_mars:
Dec 04 13:51:48 raspberrypi mariner[349]:   File "/opt/venvs/mariner3d/lib/python3.9/site-packages/mariner/mars.py", line 50, in __enter__
Dec 04 13:51:48 raspberrypi mariner[349]:     self.open()
Dec 04 13:51:48 raspberrypi mariner[349]:   File "/opt/venvs/mariner3d/lib/python3.9/site-packages/mariner/mars.py", line 44, in open
Dec 04 13:51:48 raspberrypi mariner[349]:     self._serial_port.open()
Dec 04 13:51:48 raspberrypi mariner[349]:   File "/opt/venvs/mariner3d/lib/python3.9/site-packages/serial/serialposix.py", line 325, in open
Dec 04 13:51:48 raspberrypi mariner[349]:     raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
Dec 04 13:51:48 raspberrypi mariner[349]: serial.serialutil.SerialException: [Errno 13] could not open port /dev/serial0: [Errno 13] Permission denied: '/dev/serial0'

jackslao, 2021-12-05:
Update:

Nice! I got it working. I had to go into 'raspi-config', **disable** login shell AND **enable** serial port hardware. They are different. I didn't know that until I thought about it for few hours...

@JMAN7102 thanks for your help!

amd989, 2026-11-11:
Closing: this was about installing on Raspbian buster with Python 3.7 from the old PPA. [Mariner 1.0.0](https://github.com/mariner3d/mariner/releases/tag/v1.0.0) ships for Raspberry Pi OS bookworm and trixie with its own Python, so these steps no longer apply. Install instructions: https://mariner.readthedocs.io/en/latest/software-setup.html

If this still happens on 1.0.0, comment here and I'll reopen.

closed 2026-11-11 by amd989