luizribeiro/mariner #490: package.yaml Action has not completed successfully since a run  on 9th September 2021

BlueFinBima, 2021-12-01:
**Description**
The package action on this project (and also the 4.4.3) fork has not been successful recently.  dh-systemd cannot be found.
**Reproduction Steps**
Steps to reproduce the behavior:
Actions->package.yaml->Build and look in the log.

**Expected Behavior**
A deb package artefact will be produced.

**Screenshots / Logs**
If applicable, attach screenshots or logs to help explain your problem.

**Details:**
```
Step 3/14 : RUN apt-get -y install sudo dpkg-dev debhelper dh-virtualenv   dh-systemd python3 python3-venv
E: Unable to locate package dh-systemd
The command '/bin/sh -c apt-get -y install sudo dpkg-dev debhelper dh-virtualenv   dh-systemd python3 python3-venv' returned a non-zero code: 100

Error: Process completed with exit code 100.
```
**Additional context**
Last successful run was 9th September 2021.

J-Iszatt, 2022-01-12:
Any updates here?
I am still struggling with a build of mariner on my pi.

@BlueFinBima: I checked some of your changes on your branch and am happy to see that you have been working on it recently! Would be so cool to see an update any time soon.
If you are willing to share some details on the build process (maybe in the discussions section) I would love test/help as good as I can

BlueFinBima, 2022-01-23:
Hi @J-Iszatt,  I have not been working on this, and no further work will be done by me until the packaging action can produce an installable deb.  I don't even have a deb for changes I created on my own fork because it was automatically deleted.  I also don't have the Linux skills to fix this packaging problem, so until some kind and skilled person steps up to fix this issue, progress is unlikely to happen.

fcollingwood, 2022-02-02:
@luizribeiro is Mariner dead?

fcollingwood, 2022-02-02:
@luizribeiro I have created a pull request to fix this issue.

BlueFinBima, 2022-04-06:
Not clear to me who/what fixed this, however a .deb is not being produced so I will close this off.  I will also attempt to port changes over to my fork.  Thanks to everyone who tried (and succeeded) to get this working.

closed 2022-04-06 by BlueFinBima