luizribeiro/mariner #492: Issues installing / launching with latest raspbian bullseye

Landondesi, 2021-12-26:
I have tried a few times now to install older version of python (3.7/3.8) but it would appear that something fails during mariner install and then again when launching that causes the Python errors "Could not find platform independent libraries" and all proceeding errors associated with such.  Will this software run on Python 3.9? Is there a quick workaround to point it to the proper installation?

antwal, 2022-01-07:
Fixed here #493

gitautas, 2022-05-15:
Recompiling python is not a fix.

antwal, 2022-05-15:
> Recompiling python is not a fix.

currently the code only works with 3.7, when it is totally updated to 3.9 the problem can be solved. Raspbian bullseye uses version 3.9 as a base

amd989, 2026-11-11:
Closing: this was about installing on Raspbian buster with Python 3.7 from the old PPA. [Mariner 1.0.0](https://github.com/mariner3d/mariner/releases/tag/v1.0.0) ships for Raspberry Pi OS bookworm and trixie with its own Python, so these steps no longer apply. Install instructions: https://mariner.readthedocs.io/en/latest/software-setup.html

If this still happens on 1.0.0, comment here and I'll reopen.

closed 2026-11-11 by amd989